iNavFlight/inav #2725: INAV 1.8.1 with Matek F722

bavariangerd, 2018-02-05:
1. After flashind the Matek F722 with the new INAV file, the board could not be recognized  from USB controller - flashing Betaflight - all works fine - flashing INAV again - same problem
2. Flashing 4 in 1 ESC with GLHeli - the ESC`s are deleted, but firmware is not flashed - always abourt between 50 and 60 %. Flashing the ESC`s with Betaflight - all works fine.
I would like to use INAV because of its GPS performance
can somebody help please
Thank you

digitalentity, 2018-02-05:
Hi. Although I can't reproduce the USB issue, I've made a small fix to init sequence.
 Could you test this hex? [inav_1.8.1_MATEKF722.zip](https://github.com/iNavFlight/inav/files/1694989/inav_1.8.1_MATEKF722.zip)

bavariangerd, 2018-02-05:
OK. I`ll test it this afternoon.
Thank`s

bavariangerd, 2018-02-05:
I testet it right now.
After flashing the board an reconnect, the blue light is blinking twice - the virtual comport
gets recognized, but i got no connection to the board.
In Betaflight the blue light of the board is blinking all the time after plugging in the USB

digitalentity, 2018-02-05:
Reflash with full erase. If the USB port is detected correctly it's either a garbage configuration remaining from Betaflight or OS driver issue.

bavariangerd, 2018-02-05:
i did it with full erase

bavariangerd, 2018-02-05:
it is the same behavior as before i flashed it with Betaflight.
First time i flashed it with INAV it works fine.
I had an Bluetooth communication on UART2 - works fine, 
but I could not connect USB anymore, so i reflashed it.
May be thats the reason why

digitalentity, 2018-02-05:
Can you describe exact steps to reproduce the issue? It just works perfectly fine for me.

bavariangerd, 2018-02-05:
I flashed the new board with INAV 1.8.1
connection with USB works. Then i made Configuration

1. UART1 = Bluetooth Modul(MSP)
2. UART2 = SBUS(RX)
3. UART3 = Telemetry(Smartport)
4. UART5 = GPS(UBLOX) Baudrate: 38400
CLI: Telemetry_inversion = ON
defined all Modes etc.
All works fine - connection with Bluetooth works fine
Then tried to flash 4in1 ESC via Bluetooth witch failed - see description above.
Tried to connect the board via USB - failed - as i plugged in USB to board, the blue
LED on the board is blinking twice - then turns off and stays OFF.
I flashed the board again with INAV 1.8.1 - same procedure - blue LED is blinking twice
and turns OFF - board was not recognized from USB controller.
Than i flashed the board with Beatflight and all works fine - Bluetooth an USB.
Flashed again with INAV - blue LED blinks twice - USB not recognized - no connection to board.
After i flashed your version inav 1.8.1 then USB controller recognized the virtual com but
the board is not communicating - blue LED blinks twice then turns OFF.
Flash with Betaflight - blue LED is blinking all the time and communication works fine

digitalentity, 2018-02-05:
@bavariangerd ok. Let's try investigating further. Can you test if USB connection works with **nothing** connected to the board. If it does - please start reconnecting things one by one until it stops working.

bavariangerd, 2018-02-06:
I did so and it works.
The reason was the compass(SDA/SCL) - after plugging in the board thers is no 
communication possible. The power supply for the GPS was 5V - I changed it to 3,3V
and now it works(why did it work in Betaflight with 5V?)
I read about this GPS, that it does not matter whether you plug in to 5V or 3,3V, because
it got a regulator on board.
So, at the end it was my mistake. 
I apologize for making so much trouble to you and
thanks a lot for helping.

bavariangerd, 2018-02-06:
Tast question(I hop)
where can i see the compass direction in INAV?

DzikuVx, 2018-02-06:
INAV Configurator, Setup tab, look for heading

closed 2018-02-21 by DzikuVx