iNavFlight/inav #2727: Naze32 rev6 drifting with inav 1.6

amitmanish, 2018-02-05:
I am facing drifing in my quad with naze32 rev 6. I am using onboard magnetometer and default inav settings. My rc channels are at 1500 with theottle at 1000.
Drone shows level in gui but drifts to right and back in acro mode.

shellixyz, 2018-02-05:
Can be a vibration issue or your gyro have been damaged. Also it is not recommanded to use on-board magnetometers.

amitmanish, 2018-02-06:
I tried using external magnetometer but got i2c error as onboard and external compass on ublox are same HMC5883.FC is tightely mounted on frame.

shellixyz, 2018-02-06:
Anyway the magnetometer has nothing to do with the drift it is only used for the YAW axis and only for navigation modes I believe.

> FC is tightely mounted on frame.
This can make vibrations an issue. Try to soft-mount your FC. You will find a lot of info on how to do it on the Web.

Please post at least your config dump. A flight log showing the issue would be nice too. Otherwise there is little we can do.

DzikuVx, 2018-04-25:
NAZE is end of life, we are no longer supporting this board I'm affraid. Plus, this looks like a hardware problem to me

closed 2018-04-25 by DzikuVx