iNavFlight/inav #2730: Feature request: vtx power on a switch or based on arming state.

P-I-Engineer, 2018-02-05:
I've heard of this coming for betaflight. Any chance it will be incorporated in INAV.

unarmed: 25mw
armed: to some set power value.

or switch controlled and i'll do it as a logical switch in my radio.

other thought would be to have dist from home control power.
with distance thresholds for power levels.

within 25m of home set to 25mw
vtx_pwr0 = 25mw
vtx_dist0 = 25m

from 25m to 1000m set to 200mw
vtx_pwr1 = 200mw
vtx_dist1 = 1000m

and so on.

Hazakins, 2018-02-05:
Would be pretty nice. Would keep my Unify pro from melting down but still be able to use 800mw

krzysztofmatula, 2018-02-06:
> other thought would be to have dist from home control power
This may be tricky... It rather depends on what kind of obstacles are on the way. Not just the distance...

malarcky, 2018-02-26:
Yes please, that would be nice, very convenient to use with smartaudio.

malarcky, 2018-05-24:
I dont have the skill to code, but cant understand how this is not made or forked from cleanflight or betaflight yet!

closed 2018-08-06 by stale[bot]