iNavFlight/inav #2733: Undocumented FS entry condition?

Hwurzburg, 2018-02-06:
Wiki states only three conditions can force FS....but I have been told on the Slack channel that its possible SBUS has a FS flag that iNAV might be using...if so, and there is no way to disable this, its a big issue for me flying nonGPS (ie no RTH) proximity with planes since there is no way to avoid a short glitch DISARMING the plane...this would result in a crash even if the signal loss is momentary since its not possible to switch the ARM switch, lower throttle, wait the FS reacquisition timeout,reARM and recover the plane if the signal is re-acquired...its crashed by then.

If this is true, there should be a way to disable this and allow instantaneous re-acquistion of signal and flyability for SBUS RX planes that do not use GPS.

in a non GPS plane its desirable to have FS (no pulses) drop the plane(disconnected or failed RX).....but not on signal loss if the RX is programmed to desired channel settings for TX loss...I normally would program: ARM, ANGLE, low throttle, and slight turn....hoping to re-aquire control....but there seems no way to do this currently

shellixyz, 2018-02-06:
The DROP failsafe procedure disarms that's what is written in the doc but you have other options. If you want to use RX FS channels config I think you should use the NONE procedure.

Hwurzburg, 2018-02-07:
thanks....that will do for my purposes, but it wont help in a rx failure or disconnect....having a setting to prevent the SBUS FS flag from triggering a FS would allow both what I want and the ability to protect against the other failures....
at the least, the SBUS FS flag condition should be documented in the FS docs...

Hwurzburg, 2018-02-07:
but there would be no pulses also, which trips FS also

Hazakins, 2018-02-07:
Why not set failsafe to "do nothing" and set your custom FS on your radio while binding your RX. Assuming things like frsky sbus RX and Taranis TX

Hwurzburg, 2018-02-07:
that was Shellixyz's comment above...after thinking about it, using LAND and adjusting the params should do it...that will also take care of the case if the RX fails or is disconnected....which NONE will not do...
am closing this....would be nice to update the wiki with FS flag in SBUS stream as being the fourth way to enter FAILSAFE

closed 2018-02-07 by Hwurzburg