iNavFlight/inav #2740: Beitian BN-880 strange behavior

Antiheld86, 2018-02-08:
Hi,

I have an Issue with my newly arrived Beitian BN-880. The Magnetometer is recognised correctly and responds to turning it. The Problem is, it does not response when I turn it horizontally but when I Pitch/Roll it and when it is turned on one side and then rolled like a wheel.

### Board and Version
Matek F722 / Beitian BN-880

INAV 1.8.1
# version
# INAV/MATEKF722 1.8.1 Jan 28 2018 / 07:30:22 (5afb5b8e)

### Behavior
* Compass is not reacting to horizontal turns but vertical turns

_Is this a new feature request?_
No

#### Blackbox logs/Config (.ZIP) where problem occurs
2018-02-08 @ 19:35:48 -- Running - OS: Windows, Chrome: 64.0.3282.140, Configurator: 1.8.1
# dump

# version
# INAV/MATEKF722 1.8.1 Jan 28 2018 / 07:30:22 (5afb5b8e)

# resources

# mixer
mixer HEX6X

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
[... 502 lines cut ...]
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 18
set pitch_rate = 18
set yaw_rate = 9
set tpa_rate = 0
set tpa_breakpoint = 1500

#

Antiheld86, 2018-02-08:
It turned out that I don't really understand the way the magnetometer works and with the correct alignment setting it works.

Sorry

Antiheld86, 2018-02-08:
I now have 90°-flip allignment and it correctly reacts to turning around yaw. But it still drifts.

closed 2018-02-08 by Antiheld86

shellixyz, 2018-02-09:
Did you do the calibration ? Also it is away from potentially interfering parts (motors, metal or carbon fiber, high power cabling) ?

Antiheld86, 2018-02-09:
Yes, i had the Calibration done. I think it were interference. It works better now. Thanks for your help!