iNavFlight/inav #2745: OSD Max Distance incorrect for long flights > 0.5 miles

pjbca, 2018-02-10:
This is a bug report.

At the end of a flight in iNav 1.8,  the summary screen ("Stats")
"Max Distance" gives an incorrect reading.
For example, in this flight:
https://www.youtube.com/edit?o=U&video_id=d9Srq-z_tLE
the pilot clearly flew 0.67 miles from home.
But the screen gives 0.35 miles in the summary.
I have observed similar in my flights, using iNav 1.8, Omnibus F4 V2 board.

Steps to reproduce this:
Fly in iNav greater than 0.5 miles from home.
Look at "Max Distance" in summary screen at end of flight.
It reads less than 0.5 miles.

My idea about how to drill down into the bug:
According to the osd.c file, the summary should only give "Max Distance" in miles if it was greater than 0.5 miles. If it is less than 0.5 miles, it should give the "Max Distance" in feet.
The fact that the stats page is displaying 0.35 miles max distance indicates that this is a bug. It should never happen.

The method/function osdFormatDistanceStr is the suspect method in the file osd.c
osdFormatDistanceStr creates the stats string from the internally stored stats.max_distance variable.
stats.max_distance keeps a running tab on the max distance the UAV has flow from home.

I suspect it is related to the fact that osdFormatDistanceStr takes a 32 bit argument, whereas stats.max_distance which stores the max distance is 16 bit. However I am unable to run a debugger on iNav so it is hard to confirm or test this. (Can any developer run a debugger and if so could they please post directions on how to configure it????)

[image]
[image]

shellixyz, 2018-02-10:
Looks like you are right there is a bug there. The issue seems to be that max_distance is using the uin16_t type and the unit is centimeters. 65535 centimeters = 0.40721561 miles which matches the max distance shown on the OSD stats screen in the video from the link you provided (broken link by the way).

shellixyz, 2018-02-11:
For debugging see [developement documentation](https://github.com/iNavFlight/inav/tree/master/docs/development)

pjbca, 2018-02-11:
It might be more subtle than that. Max_distance is stored in meters. But it is passed in cm to osdFormatDistanceStr. 

Sorry about the link broken (not sure why screenshots don't show, I uploaded them to github comments), try this:
https://www.youtube.com/watch?v=sEWd6nbj8Zo

To debug subtle things like this you need an IDE with line by line code execution and the ability to inspect the value of each variable as the code executes. I was able to build iNav using various methods in the documentation, but the IDE Eclipse debug setup instructions seem outdated and it do not work for me. I am willing to use any IDE or any OS. Just let me know what is working for the developers and I will mirror their setup.

digitalentity, 2018-02-12:
@pjbca possibility to output debug information to serial port was just added to `development` branch (https://github.com/iNavFlight/inav/pull/2691). You can use it to periodically output run-time debug information from arbitrary places in the code.

shellixyz, 2018-02-12:
@pjbca I'm seeing the screenshots no issue. The max distance is passed in cm to osdFormatDistanceStr but in an int32_t and the compiler should automatically cast `stats.max_distance*100` to 32bits it is probably not the issue.

fiam, 2018-02-25:
Managed to reproduce this one in the bench. I'll submit a fix tomorrow.

digitalentity, 2018-02-26:
Fix merged. Closing.

commit f9c034bd (2018-02-26): Fix calculation for miles in osdFormatDistanceStr()

Reported incorrect numbers in maximum and total travelled distances
in the OSD.

Fixes #2745
closed 2018-02-26 by digitalentity