iNavFlight/inav #2746: SPRacing F3 - Unable to make SmartPort Telemetry work over SoftSerial

Rups63, 2018-02-11:
### Board and Version
SPRacing F3
iNav 1.8.0

### Behavior
I've been unable to make SmartPort telemetry work of SoftSerial on my X6R.

The X6R is bound channels 1-8 With Telemetry.  I have SBUS going to UART3 on the SPRacing F3 board and GPS going to UART2.  In the Configuration tab in iNav, I have 'Telemetry Output' enabled and I've also enabled 'Enable CPU based serial ports'.

In the Ports tab, I have Softserial2 set to SmartPort at 57600.  I have pins 5 and 6 from IO2 on the SPRacing F3 board both connected to the SmartPort output on the X6R receiver.

I know that telemetry is working on the X6R receiver, because when I 'Discover New Sensors' the RSSI and the Rx Voltage show up.  But, none of the other telemetry data from the SPRacing F3 board shows up.

I have tried leaving the baud rate set to 'AUTO' in the Ports tab for the SmartPort telemetry, but that didn't seem to matter.  I also tried switching the SmartPort telemetry to IO1, using pins 5 and 6 from IO1 to the SmartPort output on the X6R, and that didn't seem to matter either.  I made sure the board was armed when I tried to 'Discover New Sensors', and that didn't matter.  I even went back to the basics and carefully redid my wiring and the crimp on connections to be sure that they were perfect.  I just can't seem to get the telemetry working.

Link to the Dump is below.  Have I set something up incorrectly?  This same board with the same setup has no problem passing telemetry in Betaflight - I use this same setup on two quads, one with an XSR receiver and the other with an X4RSB receiver.

#### DUMP
https://pastebin.com/JvYFj4f7

Rups63, 2018-02-11:
Edit to add that I also tried another X6R receiver I had in the drawer.  Same results as above.

Rups63, 2018-02-11:
I also tried an X8R and an X4RSB - same result - Softserial isn't working.  So this can't be a receiver issue.

shellixyz, 2018-02-11:
Looks like you did almost everything right but you should not connect the RX line of the softserial port (only TX). To make it work you need only to:

- Enable the SOFTSERIAL feature
- Enable the TELEMETRY feature
- Set telemetry type to smartport for the serial port you are using in the ports tab
- Connect the receiver Smartport wire to the TX line (only) of the FC. In your case IO_1/6(CH6) for SOFTSERIAL1 or IO_2/6(CH8) for SOFTSERIAL2.
- Make sure your receiver is powered on at the same time as the FC is.

No need to change the baud rate it is fixed it will have no effect. No need to arm the board to receive the sensors.

Rups63, 2018-02-11:
I'll give this a shot later - I hadn't thought to try it.  Makes sense - we're not after a bidirectional connection like I'm after with Betaflight.

Rups63, 2018-02-11:
I tried using IO2/6 only on Softserial2 for the SmartPort telemetry.  Telemetry and Softserial were both enabled.  It did not work.  I tried IO2/5 only on Softserial2 as well, in case I wasn't sure which was TX or RX - that didn't work either.

Using UART the telemetry goes through - but the vehicle has to be armed.

shellixyz, 2018-02-11:
That's odd. S.Port is working for me with an OMNIBUS F3 on softserial. I don't remember if it was v1.8 or the dev branch though. It should work on the SP RACING F3 I checked the SS definitions for this target and everything seems to be alright. I definitely didn't have to arm to receive the telemetry data.

Rups63, 2018-02-11:
I use Omnibus F3 boards with iNav as well.  I've never had to use Softserial on the Omnibus boards because I've had enough UARTs.  On this SPRacing F3 board, there are less open UARTS, so I had to try and use Softserial.  And like I said before, on Betaflight everything works without a hitch.  Luckily I can use UART1 on this SPRacing board to get telemetry.  Softserial just won't cooperate - on either IO1 or IO2.

bearotor, 2018-02-12:
Your wiring might be the problem. I've setup telemetry on my SPRacingF3 using SS2 with a resitor between rx and tx like this:

```
CH5 ------------------- + -------- SmartPort
			|
CH6 ----- [10kOhm] ---- +
```

Rups63, 2018-02-12:
I've not tried the resistor. I have the telemetry working on UART1, and wont be rippong imro the guts of the vehicle anymore. It's interesting that the same board, running Betaflight, works the way I had Soft serial wired up, but not with iNav.

OlivierC-FR, 2018-02-18:
Just saying, I have a SP Racing F3 ("banggood clone") talking fine to my R9 receiver on one of my plane.
I do get all telemetry values on my X12s telemetry screens and log file.
With iNav 1.8, softserial 1, pins 5 and 6  together as detailled above, port speed 57k.

mueller2030, 2018-03-08:
Hey guys, found this thread cus I was working on the same issue. Since this fc has no osd build in I really wanted some vbat telemetry I had tried both soft serial ports and could not get it to work just like mentioned above.
So now I have it wired on UART 1 and telemetry works only when armed like mentioned above.
I have nothing more to say but just wanted it to be known that this does seen to be a common issue.
Thanks

closed 2018-04-25 by DzikuVx