iNavFlight/inav #2761: Question

wookie666, 2018-02-13:
So sorry if this is the wrong place. I just have tried everything and cannot find information

Board is Betaflight F4
Trying to see if there is a way to resource i2c so i can run a compass with GPS. 

Not sure if it can even be done.

Anyone know about this

Thanks.
Chris

shellixyz, 2018-02-13:
You can't remap resources at run-time. Won't be useful anyway, the I2C pins are shared with UART3 and there is nothing you can do about it. I think the only way you can have separate I2C pins to free UART3 would be to use software I2C and I'm not sure it is an available option.

wookie666, 2018-02-13:
Nope that's fine. So if they're shared on that board with uartv3. Do I just connect the scl and sda wires to uart 3 tx and Rx  and if so which one to where. Also sorry for the questions but what's the required setting in the ports tab to enable too. Thanks again

shellixyz, 2018-02-13:
SCL on TX and SDA on RX. You need to disable everything on UART3 in the ports tab.

wookie666, 2018-02-13:
Thank you very much, this might just make the betaflight f4 board a semi
good choice. Even if it doesn't have a barometer.