iNavFlight/inav #2765: Wrong formula for current meter scale

FranzAr, 2018-02-13:
The given formula here seems to be wrong:
https://github.com/iNavFlight/inav/blob/master/docs/Battery.md#tuning-using-battery-charger-measurement

It says:
current_meter_scale = (charging_data_mAh / reported_draw_mAh) * old_current_meter_scale

From my understanding it should be:
current_meter_scale = (reported_draw_mAh / charging_data_mAh) * old_current_meter_scale

Or am I totally wrong?

shellixyz, 2018-02-13:
I think you are right. if `charging_data_mAh / reported_draw_mAh` > 1 it means the measured current is less than the real current. Since increasing the current meter scale is reducing the measured current it is backwards.

digitalentity, 2018-02-14:
Fixed

PR #2766 (2018-02-14): Fix battery docs current tuning formula
Fixes #2765
closed 2018-02-14 by digitalentity