iNavFlight/inav #2771: 3d hold toilet bowl effect round 2

autodog2, 2018-02-14:
ok just a freq thing
rth and 3d rock solid hold (good thing). Did good cals. I fixed the fly aways, yep set align_mag = cw270_flip.
problem: my hold is so strong that the drone will brake to an angle of 45deg. Very hard braking effect.
Problem 2: toilet bowl effect when only pointed at east. Walk up and push it around if its pointed N,S, or West. You ask how well do it hold, 10mph gust and i can hang things on it for 10 min flight time.
This is my first build so im new to it all (be kind). Frame Realacc real1s. FC Matec f722. ESC Racer star 4in1 35A (spec beli600). Racer star 2307 fire ed. e-flit 4S lipo. Standered GPS PIDs

digitalentity, 2018-02-14:
There is something generating magnetic interference on your frame. This causes compass to provide incorrect readings at ceartain drone heading. Try moving your compass sensor further away from the frame. Bear in mind that steel parts, buzzer, motors, ESCs, battery, power wires - all provide magnetic interference.

autodog2, 2018-02-14:
ill have to extend my gps stand. Its about 4 inches off top
[image]

digitalentity, 2018-02-15:
4" should be far enough. Maybe there's something magnetic in the stand itself, maybe a screw. Also try recalibrating the compass.

autodog2, 2018-02-16:
nub question; I have been calberating compas just by rotating  like crazy. Inav is happy after, should i rotate for compas cal same as i would the acc cal steps? ex top, top down, nose down, left side down, stern down, right side down?

closed 2018-02-16 by autodog2

shellixyz, 2018-02-16:
Each axis must be aligned at least one time with the magnetic north so the best is to rotate it 360° with each axis parallel to the ground.

autodog2, 2018-02-18:
sorry for not getting back sooner. Spun a prop off while testing, broke GPS, and VTX. Wish I filmed it, 3 roles, and hit the ground all in under a second. I thank I'm going to buy a 3s, for refine tests! LMAO cant disarm fast enough. I will fill youll in if I win.