iNavFlight/inav #2774: Parse Error when trying to change Receiver Channel Mapping

Toklandarr, 2018-02-14:
### Board and Version
OMNIBUS F3 AIO from Banggood.
INAV/OMNIBUS 1.8.0 Nov  1 2017 / 06:14:28 (912d1315)

### Behavior
tried to change Channel Mapping to be able to use the 3 PWM Outputs on my SBUS Receiver for some servos directly. 
So when I wanted to map TRAE to other channels I first tried with the "GUI" under the "Receiver" Tab, but it would just always change back to the default.
Then tried CLI:

```
# map 12345678
Parse error
# map 1234TRAE
Parse error
# map 12TRAE78
Parse error
# map 1AETR678
Parse error
```
only the default appears to work...
```
# map AETR5678
Map: AETR5678
```

shellixyz, 2018-02-15:
The letter channels must always be first meaning you can only use the first 4 channels for Aileron/Elevator/Throttle/Rudder. Only XXXX5678 is valid.

closed 2018-02-15 by DzikuVx

Toklandarr, 2018-02-15:
thought it was possible after reading https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20OMNIBUS.md

> The popular x4rsb for example outputs channels 1,2,3 as PWM in addition to SBUS. Since they are shared with the channels on SBUS you need to change the channel mapping to 123AETR4and ignore the first 3 AUX channels within Cleanflight.

DzikuVx, 2018-02-15:
That part of DOCs makes no sense are refers to Radio configuration not INAV. I'm just changing it to something understandable

fugounashi, 2019-02-11:
> The letter channels must always be first meaning you can only use the first 4 channels for Aileron/Elevator/Throttle/Rudder. Only XXXX5678 is valid.

Are there any plans to support AETR outside of channels 1234, or any reason why support couldn't be added?  At the moment the 3 PWM outputs on a X4R-SB can't be used as they are fixed to channels 123.

shellixyz, 2019-02-11:
There isn't. Can't you bind your receiver to use the highest 8 channels 9-16 ? Then the outputs would be 9, 10, 11.