iNavFlight/inav #2775: Turning on LED's causes Matek F722 to Hang

jmillerfo, 2018-02-15:
### Board and Version
Matek F722-STD
http://www.mateksys.com/?portfolio=f722-std#tab-id-5

1.8.1 Feb 10 Release Candidate downloaded from MATEK site:
INAV/MATEKF722 1.8.1 Feb 10 2018 17:16:11 (c31acda)

### Behavior
After loading the above firmware, the quad copter operates normally until the LED are selected on via a Aux Mode.  The LEDs illuminate on Aux Switch but will not turn off when commanded by the same switch.  The quad will not respond to Transmitter commands, USB, OSD commands,  or Bluetooth telemetry.  

Powering off/on the Quad successfully returns the Quad to normal operation as long as the LED's remain off.  This is different behavior compared with the initial RC of 1.8.1 which would require reloading firmware to return the system to operation.

The version of 1.8.1 that appeared in #2686 works properly. 

_Is this a new feature request?_ No

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here. 

https://pastebin.com/QNxkMVC3

_Upload a zipped blackbox log (if available)_

MrAlfabet, 2018-02-17:
Check out this issue. Solved it for me.

https://github.com/iNavFlight/inav/issues/2686

jmillerfo, 2018-02-17:
I agree that the file that can be downloaded from #2686 fixes the problem.  Matek released a version on their web site subsequently dated Feb 10 that exhibits the behavior above.  Since this is after the posting of the replacement file in #2686 I wanted to alert whoever is building the Hex file that they have re-introduced another problem.

MrAlfabet, 2018-02-17:
I don't think Matek is building from the dev branch.

dleask, 2018-02-28:
The 1.9.0-RC builds now fix the issue.

closed 2018-03-01 by digitalentity