iNavFlight/inav #2777: SERVO AUTOTRIM improvements

giacomo892, 2018-02-15:
We should put in place some sanity checks to the autotrim mode and improvements like:

- Limit autotrim action in a certain range (us)
- Leave the autotrim without making any changes if the flight mode is ANGLE or HORIZON and pilot touches the roll/pitch stick. In this case pilot will cause a wrong trim
- Compute somehow a midpoint measurement noise and check if the auto trim can be considered good or not.
-Rethink the whole autotrim system avoiding to write AUTOTRIM on the OSD after then trim as being completed (aka avoiding to keep the mode enabled for the whole flight after this have being used). For this I propose to invert the logic compared to what it is now. Like if you want to keep the current trim you have like 2-3 seconds to disable the mode after the trim has being done, otherwise the midpoints will be reverted to previous ones immediately.

Hazakins, 2018-02-15:
Also quality of life idea. No modification of trim values when in ANGLE mode but it you bump the AUTOTRIM switch it would give the servo values in the SYSTEM MESSAGE on the OSD. In my case the wing flys perfect flat in angle mode even before I tuned the values. To be able to check the DVR and see them and use for tuning would be nice.

giacomo892, 2018-02-15:
I personally trim my wing in ANGLE mode. When it is flying leveled and there is no wind I take the occasion of trimming it on the fly. Trimming it in PASS to me is too difficult.

Hazakins, 2018-02-15:
Yes, show current servo outputs to OSD once auto trim is done for DVR capture.
Its a bit of a use case scenario, but playing with different battery packs and different cruise throttle positions. What do you think?

giacomo892, 2018-02-15:
It can be done. It is a brainstorming issue

closed 2021-06-20 by DzikuVx