iNavFlight/inav #2779: Using iNav for Model Rocket FIN control

lucasabb, 2018-02-15:
Hi there,

My frist time here so I am not used to the system if I do something wrong I am sorry.

I intend to use a flight controller 10DOF like naze32 full or SP F3 deluxe to do movable Fin control in my model rocket in order to make it self correct on itself and I would like to use the iNav do to so and I would like to know how would that work, would I be able to do it using the regular GUI  with the right PID or would I have to do some tweaks on the code? is the code open source to it? which language is it written?

tks for the help

DzikuVx, 2018-02-15:
Stabilization should work just like for an airplane. Some tweaks should be required like simulating high throttle during liftoff for everything to start working but it should be doable without coding. The only question are PID gains for it. 

But for guidance, that will rather not work, ballistic mode is not implemented

lucasabb, 2018-02-15:
First of all thank you very much,

I need 3 things.. stabilization, telemetry and some signal to start the parachute, I believe there is suport for telemetry normally used with the RC, but could I send the signal to like an Arduino?

Is there anyway I can send the signal for the chute when I reach negative speed in y axes?

thank you

digitalentity, 2018-02-16:
A very interesting task you have. I see one potential problem - extreme acceleration of model rocket. Our accelerometers are only able to handle 8G while model rocket might exceed that. As @DzikuVx noted - PIDs are another story. You can probably calculate required PIDs to be in the ballpark, but it definitely will require some trial and error to fine-tune.

digitalentity, 2018-02-16:
Regarding the chute - you'll have to code that part yourself. Or, alternatively, you can send your telemetry stream to arduino that will make decisions based on that.

lucasabb, 2018-02-16:
My first thought was to do with Arduino but I didnt wanted to embed one in the rocket just to do that. So to code it, how would have to do that? can do it using the iNav plataform? what language does it use? C?

And about the telemetry, all the tutorials I saw it was to send the telemetry to an specific rc transmitter/reciever combo but in my case the transmitter isnt really necessary and they r very expensive, how would it work to send to an arduino? can I plug a radio transmitter in one of the ports? 

Thanks for the feedback in the PID

digitalentity, 2018-02-16:
INAV is GPLv3 and written in C, so it shouldn't be any issues with modifying it to your liking.

Re telemetry. INAV can generate variaous kinds of telemetry, some of them are not intended to go to your transmitter (LTM, MAVLink). You'll probably need a transmitter (or seriously hack the code) to arm the board. Arming means that INAV would be able to fully control everything. However you can't arm in advance because you'll have PID windup and your rocket will fly anywhere but up.

I suggest implementing some code that will detect a takeoff and arm the board. Alternatively you can have INAV ignite your engine on arming.

lucasabb, 2018-02-16:
Great, I will look deeper into those things.

Thank you very much for the help.

digitalentity, 2018-02-16:
@lucasabb you're also welcome to join INAV's Slack https://inavflight.signup.team

NCChris, 2018-02-21:
I wouldn't us a flight controller for this application. You can do it with an arduino and an MPU6050. Two fixed fins and two fins on servos. Here is a self balancing robot with the same concept: https://github.com/jjrobots/B-ROBOT_EVO2

FPVZaphod, 2018-03-07:
For the chute release you can use the Baro Reading. Loosing altitude can release the chute, so you won't have an accidential release in the case of wrong gyro/acc interpretation. 
Maybe you should monitor the acc readings in a testflight without stabilization. Just to get an idea about the G forces?

closed 2018-05-27 by stale[bot]