iNavFlight/inav #2780: hi everybody .

tgreif, 2018-02-15:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

fiam, 2018-02-15:
Hi @tgreif. Having a nice day, mate?

P.S. The slack group is **[==> this way](https://inavflight.signup.team/)**

closed 2018-02-15 by DzikuVx

tgreif, 2018-02-15:
sorry was a mistake