iNavFlight/inav #2781: Accelerometer calibrating problem

tgreif, 2018-02-15:
Sp Racing F3 "LHI Official Store"
INAV 1.8.0 /Chrome: 61.0.3163.100, Configurator: 1.8.0

### Behavior
hi everybody 
i've installed Inav to my new SP racing F3 controller, with external OLED display Barometer (i2C)
magnetometer (i2C) and gps (Seriale) everything is working well . but i'm not able to calibrate my Accelerometer (sure i'm using the 6 point calibration) . 
I've tryed so many times in different conditions. 

Please give me a hint where I can make it work 
Is this a new feature request?_

shellixyz, 2018-02-15:
Make sure you calibrate the accelerometer relative to the board not relative to the model. The arrow in [accelerometer-calibration-steps](https://github.com/iNavFlight/inav/wiki/Sensor-calibration#accelerometer-calibration-steps) represents the arrow on the board not the front of the model.

tgreif, 2018-02-15:
hi shellixyz. thanks for response but i think that i did like this . 
[image]
where yhe arrow is the direction of arrow on FC. 
"top,bottom,left up right down.

giacomo892, 2018-02-15:
As soon you did 1 and 2 you can randomly do the other 4 steps and it should work ok.

tgreif, 2018-02-15:
yes it should but  unfurtunatly it does not, 
thanks for response . 
i'm hoping in the new version if inav  there is an tool.

shellixyz, 2018-02-15:
What do you mean by "the calibration doesn't work" ? You get a red cross in the configurator next to "Accelerometer is calibrated" ? In the next version of the configurator I think there will be a guide to indicate each step but the procedure will stay the same. After the calibration what are the values of the `acczero_*` and `accgain_*` settings ?

tgreif, 2018-02-15:
hi , 
i mean the red cross  in the configurator,, 
[image]
accgain 4096-
[image]
instead acczero is 32760
[image]

tgreif, 2018-02-15:
the  cleanflight and betafliegt (simple calibration work)

tgreif, 2018-02-15:
i send only the first fotos , top and bottom for limit the data 
[image]
[image]

shellixyz, 2018-02-15:
You can check if your accelerometer is working correctly. In the configurator go in the sensors tab and look at the accelerometer curves. When your FC is flat the green and blue lines should be near 0. When on the side the green and red lines should be near 0. And when on the front the blue and red lines should be 0.

tgreif, 2018-02-15:
thanks . 
i think there is an problem on it 
[image]
blue line is always under 0 between -3 and -4 depending on the position. 
so i will try to change it , i have an external board with the 6050 . i will connect to i2c . 
do i need to cut the original one ?? what pin?

shellixyz, 2018-02-15:
Indeed that would explain why the 3 axes calibration does not work. You probably can use an external board with a 6050 MPU but not without modifying the source code I'm afraid. If you have the tools and the skills to do it the best would be to replace the defective MPU.

tgreif, 2018-02-15:
thank , i think that i will do the next days in my lab. 
WHAT DO YOY THING WHAT IS THE BEST FC for inav ?

shellixyz, 2018-02-15:
The best one is the one suited to your project. It depends what you want to do with it. What sensors you need, how many UARTs, how many outputs, the maximum size... I personally often use Omnibus boards with F3 or F4 MCUs.

junwoo091400, 2018-02-22:
Just Maybe related to this??

https://github.com/iNavFlight/inav/issues/2806

shellixyz, 2018-02-22:
I was asking myself the same thing but I'm not sure the PR for the fix changes the way the accelerometer data is read. His issue was clearly that the data for the X axis was wrong.

tgreif, 2018-02-22:
hi everybody . thanks a lot to all. 
i 've tried a cc3d and everything is working well. 
today is arrived my new sp racing f3 . only 3 minutes to set up and calibrate. 
i will change or disable the  broken Accelerometer on the other fc . 
now i will first do everything with my new one . .
today i try to connect my Bluetooth on the serial for to connect with the android device . 
bye

closed 2018-02-22 by tgreif

DavidFpv, 2020-10-17:
hello,
I'm trying to calibrate my flight controller and ran into a problem.
When I point it to the right the calibration is done as if the board is pointing down.

[image]

Sedgee, 2020-12-23:
> 
> 
> hello,
> I'm trying to calibrate my flight controller and ran into a problem.
> When I point it to the right the calibration is done as if the board is pointing down.
> 
> [image]

Hi David,
Did you manage to resolve this?
I'm getting the exact same problem with a GEPRC SPAN F722 HD BT FC board using INAV 2.6. The board is aligned correctly with the arrow on top pointing forward.
I have corrected 90deg yaw which makes the graphic in Setup page correct but when I go into sensors it shows that the orientation is out by 90deg.
If I set yaw back to 0deg then the Sensors page shows it working correctly but the Setup page graphic is out 90deg on yaw.
Any help would be appreciated.