iNavFlight/inav #2782: LED_STRIP lock up on F4 compiled with GCC 7

stronnag, 2018-02-16:
Continuation of #2372.

It has now been demonstrated that F4 hex for at least Matek F405OSD and Omnibus F4 Pro V3 that enabling LED_STRIP causes the board to lock up during initialisation (verified with gcc 7.2 and 7.3). Hex compiled with gcc 6.x are not affected.

digitalentity, 2018-02-17:
This is a real pity. Any idea what might be causing this?

digitalentity, 2018-02-18:
I'm going to have a crack at this bug. I have a gut feeling that it might be a buffer overflow issue yet again.

digitalentity, 2018-02-18:
Nailed it. Hardfault was caused by division by zero in ledstrip code. For some reason the same code didn't yield a hardfault exception under GCC 6 and earlier.

Interestingly `DIV_0_TRP` exception is masked in both cases, so division by zero should yield zero and not generate a fault. Maybe analyzing assembly listing may provide some insight but I'm too lazy to do it 😄

stronnag, 2018-02-18:
good job. Always satisfying nailing this sort of problem.

digitalentity, 2018-02-18:
And always amazingly hard to track.

digitalentity, 2018-02-18:
Fix merged.

closed 2018-02-18 by digitalentity