iNavFlight/inav #2786: Compiling builds after merge of #2748 now showing warning messages

MRC3742, 2018-02-17:
I'm not sure if this is a problem, but since the merge of #2748 Add protection against unconnected VCP , a warning message appears part way through the build process. This is showing up in all builds I have made since that pull request was merged.

Building SPRACINGF3EVO

./src/main/drivers/serial.c: In function 'serialIsConnected':
./src/main/drivers/serial.c:109:39: warning: passing argument 1 of 'instance->vTable->isConnected' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
         instance->vTable->isConnected(instance);
                                       ^~~~~~~~
./src/main/drivers/serial.c:109:39: note: expected 'serialPort_t * {aka struct serialPort_s *}' but argument is of type 'const serialPort_t * {aka const struct serialPort_s *}'

This is also showing up in all The Travis CI builds listed since that date.
The build completes successfully and seems to work properly. 
If this is not an issue  please comment and close this issue

fiam, 2018-02-17:
@MRC3742 Thanks for noticing the problem. I've just submitted a fix in #2787

digitalentity, 2018-02-18:
Fixed

PR #2787 (2018-02-18): Make the isConnected() function in the serial vtable accept a const arg
Fixes warning during build.
Fixes #2786
commit 06c62fc9 (2018-02-18): Make the isConnected() function in the serial vtable accept a const arg

Fixes warning during build.
Fixes #2786
closed 2018-02-18 by digitalentity