iNavFlight/inav #2788: Feature idea : New failsafe mode for multirotors : CLIMB (just climb, nothing else)

OlivierC-FR, 2018-02-18:
After reading this : https://github.com/iNavFlight/inav/issues/2688 (Multirotor RTH using GPS heading)

Since  it's obviously not easy to implement mag-less RTH for small multirotors, what about an additional failsafe mode, when the copter would just climb straight ?

Upon failsafe condition, stabilize the quad, and climb to target altitude, the one set for normal RTH ?
Watching my own and my friends videos, the number one cause for FPV video loss on a miniquad is by far the pilot going too low, and/or behind a large object (building, cliff, hill, trees). Against this the pilots are now almost all reacting the same : switch to angle/horizon mode, and throttle up until you get the video back and are then able to manually pilot the aircraft again. I could link to a dozen of videos where a quad was saved by going full throttle in the blind and waiting for the video feed to come back.

iNav dong this automatically would be neat.

giacomo892, 2018-02-18:
Real issue is for how long to climb and what to do if RC link is not re established after the climb. Dropping a quad from that high is not really a good idea.

OlivierC-FR, 2018-02-18:
The quad climbing ANY distance is better than nothing, and falling from 100 or 200 meters high won't make any difference. As a reminder, here talking about miniquads usually flown in acro mode, not stable camera cruisers already in angle mode.
Would not even need a GPS.

Case samples : 
https://youtu.be/JbBMXF43-K0?t=226
https://youtu.be/TBxYbjY1G2g?t=272
https://youtu.be/Zw5R4UTxAsU?t=234

digitalentity, 2018-02-18:
Set failsafe procedure to `LAND` and set failsafe_throttle above hover. This combination will result in climbing untill battery dies.

OlivierC-FR, 2018-02-18:
Well erm why not, but if it could only climb a bit instead of rocketing to the moon... :)

giacomo892, 2018-02-18:
@OlivierC-FR I think you can use `failsafe_off_delay` parameter to set for how long you want to climb in this case and then turn engines off.

closed 2018-05-27 by stale[bot]