iNavFlight/inav #2791: Feature Request

linuxslate, 2018-02-18:
New Feature Request:

Add Call Sign Display to OSD.

Note that (at least in the United States, but I believe the laws are similar elsewhere) Ham radio operators are required to identify the station at the beginning of transmission and every 5 minutes during the transmission.

The exception (again, US) would be FPV transmitters with 25mW or less, which may operate unlicensed. but I think it is safe to assume that very few iNav users would be running <25mW VTX's.

Also, if the camera itself has an OSD, the Call Sign may be able to be programmed into that, but in the case of iNav, that presents some problems:
1. Not all cameras have this capability.
2. It shares display area with the OSD.
3. The camera OSD's cannot just display the Call Sign periodically.

This is very easy to implement in FPV where a OSD is in use. At least some mods of MinimOSD did this, so the code is out there.

Note that this could be presented as more than just a feature request, since technically it is illegal to operate iNav/OSD without displaying your station identification.  (A card with the call sign could/should be held in front of the camera prior to flight, but then the flight is limited to less than 5 (10?) minutes.)

Another legal way to do this would be to send the call sign as an audio recording/digital voice/morse code over a VTx audio channel, but I don't know of an easy way to implement that.

shellixyz, 2018-02-18:
It is already possible but it was disabled before the 1.9.1 configurator release you had to enable it in the CLI. In the configuration tab set `Craft name` to your call sign then enable the OSD item of the same name in the OSD tab.

digitalentity, 2018-02-18:
INAV has a possibility to display a craft name. This feature was enabled in Configurator 1.9.1 (released yesterday). I believe this is almost the feature you want.

linuxslate, 2018-02-19:
Thank you.  I'll try the new configurator, and try to set my call sign. With the above information it is OK to close this.

Arvycka, 2018-02-20:
I must be blind as I can't find where to set the name...
[image]

shellixyz, 2018-02-20:
Should be there just under 3D. You are probably using an iNav version that doesn't support `Craft name`. What version are you using ?

Arvycka, 2018-02-21:
1.8.0 (omnibus) with 1.9.1 configurator.

digitalentity, 2018-02-21:
@Arvycka you need firmware 1.9.0 or later

fiam, 2018-03-06:
Version 1.9 does support aircraft name as an OSD element, so I'm closing this one.

closed 2018-03-06 by fiam