iNavFlight/inav #2792: Feature Request: Auto RTH depending in battery level and distance

b14ckyy, 2018-02-18:
New Feature Request:

Did not find such a request through search so I'll open one. 

What about an Auto RTH function for bigger camera quadcopter or planes?
You could take the configured default cruise speed and home distance to calculate the estimated time to get back to home point. With the battery capacity remaining (measured by making a 10s average of battery voltage) and battery usage during the flight, you could estimate the flight time left. If remaining flight time is close to the return time in cruise speed, you could then give an OSD warning when going too far away to return or even trigger automatic RTH.

Calculation example for my Camera quadcopter:

5000mAh and flight time 20min. 
flying a mission and the copter is far away
looses about 0.1V every 2min (per cell)
cruise speed is 5m/s or 18km/h
if the copter is 500m away, it should trigger a warning at 3.5V and auto RTH at 3.4V to land at home point before reaching 3.3V per cell
if the copter is 1000m away the warning and RTH should trigger at 3.6 and 3.5V and so on. 

You could also implement this together with wind speed estimation #2450 to compensate the additional power drain when flying against the wind.

shellixyz, 2018-02-19:
I planned to do something like that. It is now possible with the calculated remaining battery capacity based on the power draw in the latest release. I thought about the RTH but I don't really like the idea of automatically triggering RTH. If it was done there should be at least way to cancel the RTH. I was planning to only add a timer with the fly time left before needing to come back.

b14ckyy, 2018-02-19:
RTH should only be an Option the user can activate and not forced. Just in case you don't have an OSD running and flying LOS and avoid emergency landing into a tree or on other objects. There are a lot of people not having telemetry, esoecially beginners.

chiaweiho, 2018-03-09:
I hope that later versions can compete with DJI; maybe INAV can replace them.
Later versions hope to increase the low voltage and electronic fence system to reach Auto RTH.

closed 2018-05-27 by stale[bot]