iNavFlight/inav #2796: SD Card doesn't work on MATEK722

StefanRvO, 2018-02-19:
I am using a MATEK F722 board with iNav 1.9.0-RC1. 

When trying to use the SD-Card for blackbox logging, i get the following error in the configurator: "Fatal error. Reboot to retry", and blackbox logging does not work during flight.

Logging works in the newest betaflight, so there should be no issues with the hardware.

dleask, 2018-02-19:
Confirmed, was also the same with 1.8.1 and both Matek provided 1.8.1 hex's
Works fine in Betadlight

StefanRvO, 2018-02-20:
I am willing to try to find the reason for the issue myself, but as I haven't worked with iNav before, it would be helpfull if someone could give some hints at what parts of the code could cause the issue.
As I guess the problem is probably some kind of misconfiguration of the MATEKF722 target, I have tried to compare the target files in iNav with the ones in betaflight, but didn't find any difference which I thought was likely to be the cause.

dleask, 2018-02-20:
I’ve tried three different SD cards today and I can get one of them to work with the 1.9.0 firmware. Looks like it may be down to sensitivity of the card format. I’ll fry different formatters to see if there is some consistency.

MrAlfabet, 2018-02-20:
I'm using the hex from this issue: https://github.com/iNavFlight/inav/issues/2686

It's working fine with blackbox logging.

StefanRvO, 2018-02-20:
I just bought a Sandisk Ultra 16GB card as suggested by the Wiki. This one seems to work as opposed to the 1 GB no-name sdcard I used before. 
However, I still don't get why it would work in betaflight and not in iNav (i tried the build suggested by MrAlfabet with the same result).

digitalentity, 2018-02-21:
It's possible that Betaflight uses lower SD-card clock.

putimir, 2018-03-09:
Same to report as StefanRvO: Sandisk Ultra 16GB HC1 works, but none of the 8 noname older 1,2,4 GB cards...
Not to much of a big deal as far as I'm concerned...

wilco1967, 2018-04-03:
Same here.....
blackbox logging doesn't work on iNav with 2 different cards (1GB + 16 GB).
Fatal error....

after Flashing with betaflight, it works... (but I'm not interested in betaflight.... ;-)

bs1957, 2018-04-29:
Can confirm the issue:

MATEK F722-STD
iNav 1.9.1
SD card: Transcend 16GB MicroSDHC Class10 UHS-1.
Reformatted with both Windows and SD Card Formatter, FAT32.
Display: "Fatal Error. Reboot to retry".
Works with Betaflight.

bs1957, 2018-05-14:
Is there any solution/advice for the issue?

wilco1967, 2018-05-15:
Tried quite a few cards. Found one that works, but others that don't.

Some of the ones not working with iNav, do work with betaflight....

Somehow iNav is really picky.... But it does work, at least with some cards

bs1957, 2018-05-15:
Could you share actual card types it works with?

wilco1967, 2018-05-15:
Also a SanDisk ultra 16 I got working

bs1957, 2018-05-20:
I can confirm that SanDisk microSDHC UHS-1 16GB works

Nicholas-81, 2018-06-21:
Здравствуйте. Подскажите, какая модель Sandisk Ultra 16GB microSDHC UHS class 1 действительно работает в Matek F722?
SDSDUNB-016G-GN3IN
SDSQUNB-016G-GN3MA
SDSQUNB-016G-GN3MN
SDSQUNS-016G-GN3MA
SDSQUNS-016G-GN3MN

Хочу приобрести, но боюсь ошибиться. Ни одна microSD карта, которая у меня есть, в контроллере не определилась.

bs1957, 2018-06-21:
SDSQUSC-016G-ANCIA
https://www.amazon.com/gp/product/B00APCMMEK/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

bs1957, 2018-06-21:
Did you try iNav? I cannot make poshold and althold work.
Thanks, Boris

Dronitoni, 2018-06-29:
I tried to use a Samsung Evo 64G, but did'nt work with my MATEK F722 STD.
Switched to Sandisc Ultra 32G, works perfect.
I'm using INAV 2.0.

closed 2018-09-11 by stale[bot]

mikko80, 2018-10-23:
Not work:
Kingston micro SDHC V30 32GB (red label)
Kingston micro SDHC 10  32GB (white label)
Kingston micro SDHC I 10 64GB (white card, blue label)

FlyingJonas, 2018-11-09:
I have issues with my Matek F722Std as well.
Cant get any of my SD-Cards to initialize.
Sandisk Ultra 16GB and Samsung EVO 32GB and some no Name Cards, all Same result in my Case...
Fatal SD Error...

expipiplus1, 2019-03-09:
@digitalentity What was the problem and fix?

DzikuVx, 2019-03-09:
@expipiplus1 SD cards work as long as good Class 10 are used. LIke Sandisk  Ultra 16GB

digitalentity, 2019-03-09:
@expipiplus1 fixed by https://github.com/iNavFlight/inav/pull/3984

NSCOTTSTEVENS, 2019-06-25:
I have a PNY 32 gig that returns the same error in INAV 2.2.0  on Matek 722se.  I formatted with Tuxera formater.  Will try windows FAT32

FlyingJonas, 2019-06-25:
It turned out that my FC Matek F722 Std was faulty. I replaced it with Matek F722 SE. Check the connections of your sd card slot. try to resolder them or replace FC. I think its hardware problem.

NSCOTTSTEVENS, 2019-06-27:
Must be...16 gig PNY also no go.  I got it locally so I didn't have to wait.  Will order recommended cards before tearing down and re-soldering PIA.  I have another FC but that will be more of a project.  Can't tune well without BB so whatever it takes.