iNavFlight/inav #2798: Recommended ARM toolchain version?

mluessi, 2018-02-19:
I would like to compile iNav from source and I'm trying to find out what toolchain version I should use. In the developer documentation there is conflicting information, e.g. for Windows it lists 4.8-2014q2, while for Ubuntu it lists 4.9.3.2014q4 and for OS-X 4.9-2015q2

If there is no recommended version, what is e.g. used to build e.g. INAV 1.9.0-RC1 that was just released?

stronnag, 2018-02-19:
gcc 6.3 or 7.3 (and in between) will work just fine, older versions may not.

mluessi, 2018-02-19:
OK, but which version is used for the hex files in the release? In my experience it's best to use the exact same toolchain version if you want 100% reproducible results..

stronnag, 2018-02-19:
I don't know, but my versions built with 7.3 work just fine.

shellixyz, 2018-02-19:
I'm using 7.1.0 with success.

mluessi, 2018-02-19:
Thanks. I was able to build it with 7.2 :). 

I noticed that the code size (flash usage) is quite a bit lower than in betaflight. E.g. for OMNIBUSF4 the flash utilization is about 90kB lower (see below). Does anyone know what the explanation for this is? I would have assumed it would be the other way around ;)

inav:
   text    data     bss     dec     hex filename
 208428   12336   42208  262972   4033c ./obj/main/inav_OMNIBUSF4.elf

betaflight:
Linking OMNIBUSF4 
   text    data     bss     dec     hex filename
 296139   10092   50016  356247   56f97 ./obj/main/betaflight_OMNIBUSF4.elf

stronnag, 2018-02-19:
betaflight does not have a functional navigation sub-system?

DzikuVx, 2018-02-21:
OK, can we close this one? I suppose so. 
AFAIK one of the reason BF code takes more are various speed optimizations. They take a lot of flash and this is why there is not GPS on F3 at all. Thanks all

closed 2018-02-21 by DzikuVx