iNavFlight/inav #2801: inav omnibus f4 v2 cant work with gps ts100 mini qmc5883l HELP

JohnWonFly, 2018-02-20:
Dear friends.
I and my friends won fly. We construct drones.
But we have one problem. 
OUR GPS ts100 mini cant works with omnibus f4 v2.
I read something about qmc5883l, but cant understand how delete magGain 185-187.
Somebody can help, to build firmware for inav omnibus f4 v2 and it works with ts100mini gps?

JohnWonFly, 2018-02-21:
just install new inav version 1.9.1
it has qmc588l driver for gps module.
its work

closed 2018-02-21 by JohnWonFly

zark0-UwU, 2021-10-13:
Hi i know this issue is closed, bu nor version 3.0.1 nor 3.0.2 seem so allow me to select QMC588L for magnetometer, and it does not work in any of the ones it allows me to select.
A bit of background: My magnetometer is the one that comes with the  board of the gps [Radiolink TS100](https://www.radiolink.com/se100), and its hooked up to my MATEK H743-wing