iNavFlight/inav #2806: F35 Lightning FC fw 1.9 cannot level after calib done 6 step

furiousfpv, 2018-02-21:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_
I'm using F35 with MPU9250 sensor in fw 1.9, after finish 6 steps, my board can not level. 
[image]

When i use FF_WING fw 1.8 and do calib in 6 step, it worked great, board level. Can you help me check again with calib process in MPU9250 sensor? 
[image]

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

digitalentity, 2018-02-21:
Will test. Are you using the Configurator 1.9.1 ?

furiousfpv, 2018-02-21:
In fw 1.9, i used 1.9.1
In fw 1.8, i used 1.8

digitalentity, 2018-02-21:
Ack. Will test and get back to you ASAP

furiousfpv, 2018-02-21:
My customers have same issue with calib process, https://drive.google.com/open?id=1-L-KMOah_carw_SIp7FWPumnAZh25kmh

Thank you so much @digitalentity

digitalentity, 2018-02-21:
@furiousfpv try doing a full erase when flashing 1.9. Helped in my case.

furiousfpv, 2018-02-21:
@digitalentity Yes, i tried many time with full erase. Same problem.

furiousfpv, 2018-02-21:
@digitalentity  BTW, i tried with FF_FORTINIF4 worked great with 6 steps calib. Board can level and all axis are good. FortiniF4 use MPU6500 library.

brat002, 2018-02-21:
I don't really know if the bug is similar, but my friend also got an very strange behaviour last week https://youtu.be/ZQlX_PnT1fE?t=36. Horizon started to behave like crazy. We seen that at least 3 times during 2 weeks. 

Matek f405-ctr
Inav 1.8.1

digitalentity, 2018-02-21:
Bug confirmed. Some other boards seem to be affected as well.

digitalentity, 2018-02-22:
Found a bug. Fix pending.

furiousfpv, 2018-02-22:
So cool, Can you give me target file of F35 for test? @digitalentity

digitalentity, 2018-02-22:
@furiousfpv sure, once I make a proper fix

furiousfpv, 2018-02-22:
Can you finish today?

digitalentity, 2018-02-22:
@furiousfpv hex for testing
[inav_1.9.0_FF_F35_LIGHTNING.zip](https://github.com/iNavFlight/inav/files/1746700/inav_1.9.0_FF_F35_LIGHTNING.zip)

furiousfpv, 2018-02-22:
[image]
It worked great @digitalentity . Thank you so much.

digitalentity, 2018-02-22:
Good to hear! Fixed.

closed 2018-02-22 by digitalentity