iNavFlight/inav #2810: Spiral of death RTH Inav 1.8

Kryptooo, 2018-02-21:
Hi,
i have a Spiral of death at RTH.
Nano Talon  with Omnibus Pro F3 V2 with Inav 1.8

Here is a Video.
[https://www.youtube.com/watch?time_continue=27&v=sznxjlBqkhc](url)

# dump all

# version
# INAV/OMNIBUS 1.8.0 Nov  1 2017 / 06:14:28 (912d1315)

# resources

# mixer
mixer CUSTOMAIRPLANE

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1506 40 -1 
servo 3 1000 2000 1521 100 -1 
servo 4 1000 2000 1471 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

smix 0 2 0 -100 0
smix 1 3 1 -85 0
smix 2 4 1 85 0
smix 3 3 2 -75 0
smix 4 4 2 -75 0

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature MOTOR_STOP
[... 645 lines cut ...]
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

# restore original profile selection
profile 1

# save configuration
save

#

shellixyz, 2018-02-21:
Does it do it every time ? The OMNIBUS FC has a SD card slot so you should consider using it to record flight logs.

krzysztofmatula, 2018-02-22:
Before trying RTH check if POS-HOLD. Is it also problematic?
Before trying POS-HOLD czeck if MAG is calibrated correctly (see inav wiki page).

Kryptooo, 2018-02-22:
The Omnibus has no Mag.
Pos-Hold same problematic.

digitalentity, 2018-02-22:
Hard to say what's going on without logs. 

Are you sure this is the right cruize throttle range for your plane?
```
set nav_fw_cruise_thr = 1500
set nav_fw_min_thr = 1300
set nav_fw_max_thr = 1800
```

Also, what's interesting. Max bank angle in automatic modes is set ot 30 deg, but aircraft clearly exceeds that in RTH. This usually happen due to one of the following reasons:

 1. Insufficient airspeed = stall
 2. Inadequate tuning

You seem to be on default PID settings. Try upping your Roll PIDs a bit - to `P = 15, I = 10, D/FF = 70` and see if it helps.

binleo, 2018-04-16:
You should set Rate Ruder lower...
I set : 
smix reset

smix 0 2 0 -100 0
smix 1 3 1 85 0
smix 2 4 1 -85 0                                                
smix 3 3 2 -40 0                            //65 to 40
smix 4 4 2 -40 0
The issue solved.........

closed 2018-06-29 by stale[bot]