iNavFlight/inav #2821: Betaflight f4 rssi pad

wookie666, 2018-02-23:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

wookie666, 2018-02-23:
Im sorry my comment didn’t come through.

I have a betaflight f4 board and trying to get the rssi working from a l9r receiver. 

I have enabled analogue rssi and tried to invert and still don’t get a valid reading. Either 0 or 99 when inverted. 

Sorry for asking this here but I can’t get an answer anywhere even from the manufacturer

shellixyz, 2018-02-23:
Please try this hex. It is a custom build using PA0 as the RSSI ADC pin.
[inav_1.9.0_BETAFLIGHTF4.hex.zip](https://github.com/iNavFlight/inav/files/1752965/inav_1.9.0_BETAFLIGHTF4.hex.zip)

wookie666, 2018-02-23:
Lifesaver. I’ll test today

wookie666, 2018-02-23:
@shellixyz is this correct. Resource is still showing A0 as being free

Entering CLI Mode, type 'exit' to return, or 'help'

# resource
IO:
----------------------
A00: FREE 
A01: FREE 
A02: FREE 
A03: FREE 
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: FREE 
A09: SERIAL1 UART TX
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A15: FLASH CS
B00: MOTOR1 OUT
B01: MOTOR2 OUT
B03: SPI_PREINIT CS
B04: BEEPER OUT
B05: LED1 OUT
B06: FREE 
B07: FREE 
B08: FREE 
B09: FREE 
B10: I2C2 SCL
B11: I2C2 SDA
B12: OSD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: ADC CH1
C02: ADC CH2
C03: FREE 
C04: MPU EXTI
C05: FREE 
C06: FREE 
C07: SERIAL6 UART RX
C08: MOTOR4 OUT
C09: MOTOR3 OUT
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: INVERTER OUT
D02: FREE

wookie666, 2018-02-23:
And there is something going on with the battery monitoring. With the same settings as before. My 3s battery shows 1.14v total. Whether i have Battery voltage or cell voltage on the OSD its both the same at 1.14v. If i adjust the scale from 110 to about 300 it does scale it up. but it should be set about 110 as per before.

stronnag, 2018-02-23:
If you're using 1.9 pre-release, you need to see the draft release notes https://github.com/iNavFlight/inav/wiki/1.9.0-Release-notes

wookie666, 2018-02-23:
thanks for that. WHat about the PA0 Mapping?

wookie666, 2018-02-23:
> RSSI still doesn't work as expected ? Please post a config dump.

No still doesn’t work. I’ll get one in a second.

wookie666, 2018-02-23:
Ok so RSSI is working. But not correctly.

It shows rssi 21 when it should be full and rssi 0 when radio is turned off.

I think it has something to do with the rssipad being connected to the
transistor and other parts too, as when my L9r is connected to the board,
it shows 0.7v as full rssi but if i unplug it from the board, it shows
3.3v. So the voltage is being dropped by the board.

Entering CLI Mode, type 'exit' to return, or 'help' # dump

# version
# INAV/BETAFLIGHTF4 1.9.0 Feb 23 2018 / 20:39:17 (506ec480)

# resources

# mixer
mixer QUADX

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1
servo 1 1000 2000 1500 100 -1
servo 2 1000 2000 1500 100 -1
servo 3 1000 2000 1500 100 -1
servo 4 1000 2000 1500 100 -1
servo 5 1000 2000 1500 100 -1
servo 6 1000 2000 1500 100 -1
servo 7 1000 2000 1500 100 -1

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TELEMETRY
feature CURRENT_METER
feature RSSI_ADC
feature BLACKBOX
feature OSD

[... 508 lines cut ...]
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 0
set tpa_breakpoint = 1500
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

#

On Sat, Feb 24, 2018 at 8:24 AM, Christopher Hemmings <chemmings60@gmail.com

shellixyz, 2018-02-23:
Nothing can be done about that software side. Does the Betaflight F4 FC really have a dedicated RSSI pad ? If it does it is possible it is simply not compatible with an analog RSSI signal.

wookie666, 2018-02-23:
it does have the pad. and it works. Im going to try and isolate it from the
first resistor to see if its that circuit that is causing the issues

shellixyz, 2018-02-23:
If you are willing to modify the board and leave only a connection from the pad to the PA0 pin it should work.

wookie666, 2018-02-23:
Yes that works. Smallest resistor that the RSSI pin is right besides needs
to be removed in order to use analog RSSI signal from an L9r. Im not sure
if its reversible. as the resistor is absolutely micro and you cannot see
it once you solder it off. But i now have 96 as a full signal when the
radio is right beside the wing.

shellixyz, 2018-02-23:
Well done. The mod I provided cannot be merged in the source code because it doesn't work with unmodified boards. You will have to build a custom version yourself when you want to upgrade. Here is the patch you will need:

```
diff --git a/src/main/target/BETAFLIGHTF4/target.h b/src/main/target/BETAFLIGHTF4/target.h
index 349aa1f5..d660003b 100755
--- a/src/main/target/BETAFLIGHTF4/target.h
+++ b/src/main/target/BETAFLIGHTF4/target.h
@@ -149,8 +149,10 @@
 #define USE_ADC
 #define ADC_CHANNEL_1_PIN                  PC1
 #define ADC_CHANNEL_2_PIN                  PC2
+#define ADC_CHANNEL_3_PIN                  PA0
 #define CURRENT_METER_ADC_CHANNEL      ADC_CHN_1
 #define VBAT_ADC_CHANNEL                   ADC_CHN_2
+#define RSSI_ADC_CHANNEL                ADC_CHN_3
 
 #define USE_LED_STRIP
 #define WS2811_PIN                      PB6
```

wookie666, 2018-02-23:
Damn. Anyway you can mod it on a switch? so it can be turned on and off in the configurator?

wookie666, 2018-02-23:
Or where can i find information on the best/quickest way to build my own version

shellixyz, 2018-02-23:
To turn what on/off with a switch ? RSSI ?

You can find guide to build you own version here: https://github.com/iNavFlight/inav/tree/master/docs/development

wookie666, 2018-02-23:
No to turn off which pin it gets it from. Either a0 or c0. Depending on
which receiver people are using it for.

shellixyz, 2018-02-23:
If people can't use analog RSSI wihtout the hardware mod it wouldn't be useful and it is not possible with the current implementation.

wookie666, 2018-02-23:
Ok thanks mate I’ll read up on how to add it to future builds for myself.
If you don’t mind I’ll try it out and see if I can do it. I might ask a
question or two. I’ll close this tomorrow.

shellixyz, 2018-02-24:
No problem, go ahead.

closed 2018-05-27 by stale[bot]