iNavFlight/inav #2822: Omnibus clone no servo output

xswein, 2018-02-23:
Hi, I have problem with servo outputs for flying wing. 

I have orderedd F3 Omnibus clone with current sensing: https://www.banggood.com/30-5-30-5mm-Betaflight-Omnibus-STM32F3-F3-Pro-Flight-Controller-Built-in-OSD-BEC-Current-sensor-for-RC-Drone-p-1156615.html

I can't figure out how to enable servo output, I succesfully configured 2 other flying wings (with F1 Naze32 and SPRacing F3 clone) both worked out of the box just selecting preset  600mm flying wing.

But this one is not working, i have done:
1. Flashed latest inav
2. Selected 600mm flying wing preset
3. calibrated accelerometer
4. configured serial on rx3, ibus (flysky)
5. enabled servo output
6. set 50hz refresh rate on motor pwm and servos

In motor tab servos are moving, I can even arm (motor spins and behaves as expected) but servos just don't recieve any signal. Not even a servo glitch on bootup (servos are definitely okay, centered by servo tester).

Hint might be in resources, it seems 4 motor outputs and no servos are expected.

MOTOR 1 B08
MOTOR 2 B09
MOTOR 3 A02
MOTOR 4 A03

I guess it should be something like:

MOTOR 1 B08
MOTOR 2 B09
SERVO 1 A02
SERVO 2 A03

In Cleanflight/Betaflight i would try resource remapping, in iNav - no idea.
Any tip what could be wrong?

Thanks
Martin

wookie666, 2018-02-23:
Resource remapping is not available yet in inav

digitalentity, 2018-02-23:
@xswein please provide a `dump` output

asturchoper, 2018-02-24:
Same problem. Only channel 1 (motor) work. I have discovered that the signals of the ailerons (channel 3 and 4) come from two auxiliary channels of my tx. Channels 7 and 8. The first time I did flash everything worked fine, but I had to flash more times and now I have no sign of the ailerons on the sticks. Any ideas?

xswein, 2018-02-25:
@digitalentity dump result
# version
# INAV/OMNIBUS 1.8.0 Nov  1 2017 / 06:14:28 (912d1315)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
[... 477 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 15
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

paulo-sousa, 2018-02-26:
@xswein, are you powering the servos rail separately using an external BEC? The servos are not powered directly from the board, and you need to provide a separate 5V power to the servos.

paulo-sousa, 2018-02-26:
@xswein, are you powering the servo rail separately?

The board does not power the servos directly from the onboard voltage regulator, and you need to provide a separate 5V source to the servos.

xswein, 2018-02-26:
@paulo-sousa Are you sure? Reciever on RX3, pin J8 is definitely powered - even from USB (it works correctly). And I'm pretty sure that i measured 5V on pins  J17-J14 in between all center (5V) and right pins (GND). And BEC is rated to 3A, one 9g servo should not take more then 300mA, so i should be well below the limit. But it was late night, I'll measure it once again just to be sure.

fiam, 2018-02-26:
@xswein @asturchoper Take a look at the motors tab in the configurator. Do the servos (bars at the right) move when you move the sticks? Note that the numbers in the right represent servo channels, NOT PIN NUMBERS. For pin/servo assignments, look at the model displayed at the top left.

paulo-sousa, 2018-02-26:
@xswein, J8 is not connected to the servo rail (J14 to J17). J8 gets its power from the board (usually to power the receiver), but the remaining outputs are not powered. 

These boards have 3 zener diodes, which allow it to be powered by 3 sources, without feeding into each other:
 - ESC_5V (connected to any one of the J14 to J17)
 - VBUS (from USB)
 - VCC_5V LDO (internal 3A regulator from the board)

Do you have the ESC delivering power to J17 (most common setup)? If not, then you can measure any of the PINs and you will see the voltage dropping when you apply any type of load (like a servo).

xswein, 2018-02-26:
@fiam Yup, as mentioned in original post - servos move in motor tab, servos are connected to PWM  3 and 4 (J15 & J14) as shown on the model in mixer and should be correct. Exactly same pinout works flawlessly on Naze32 F1 and Spracing F3 on my other models.

xswein, 2018-02-26:
@paulo-sousa Nope I have multicopter ESC without BEC. May It may be that internal resistance of my multimeter is so low that current from zener diode makes multimeter read steady 5V?

fiam, 2018-02-26:
@xswein Just to be clear, both servos (bars at the right) and motors (bars at the left) are moving or just motors?

Regarding power, I can tell you for sure that the OMNIBUS F4 PRO does have the diodes mentioned by @paulo-sousa and the OMNIBUS F3 (non-PRO) doesn't have them. I don't have the exact board you linked, but from the pictures it looks like it has the diodes.

In that case, you need to provide servo power separately to any of the PWM outputs 5V pin. If your ESC doesn't have a BEC, you can use a separate one. Those 3A 5V BECs that you can find on BG or AliExpress are small and should work fine in most cases.

xswein, 2018-02-27:
Solved, it was the diode, there was false 5V without load on all 5V pins (due to not that high resistance of diode in reverse state) and even continuity in one direction (diode in forward state which i accidently tested). So i just jumped 5V from J8 to the J17 and everything works as expected. Really silly mistake. Thanks for helping guys :)

closed 2018-02-27 by xswein

benoit35690, 2018-05-29:
hello xswein, could you please give more information ? I am facing same issue.
I am powering the servo with a separate power source, not from J17.
In the ressources I see 4 motors and no servo. 
wookie666 said "Resource remapping is not available yet in inav"
Did you make something on iNav configuration ?
Thanks

xswein, 2018-05-29:
Hi, in my case, to power servos i bridged J8 5V and J17 5V pins and that solved the problem.
Some say its bit risky, single BEC is powering FC and servos, so servos overload may brownout flightcontroller mid flight. But it never happend to me - my RX sends telemetry back to TX and I have alarm set to 4.8V and voltage never dropped below - so I think I'm safe.

In my case resources mapping is not related to powering the servos, there was simply no power on the servo pins.

Resources (according to my limited experience) are mapped depending on mixer, motors first, then servos.

Preset mixers and their resources maps are here: https://github.com/iNavFlight/inav/blob/master/docs/Mixer.md

As a custom mix I experimented with VTOL flying wing (tailsitting orientation is level for easy hovering). These were my mixes:
# mixer
mixer CUSTOMAIRPLANE
mmix 0  1.000  1.000  0.000  0.000
mmix 1  1.000 -1.000  0.000  0.000

# servo mix
smix 0 2 1 -100 0
smix 1 3 1 -100 0
smix 2 2 2 -100 0
smix 3 3 2 100 0

Resources were assigned:
PWM1 -> M1
PWM2 -> M2
PWM3 -> SERVO1
PWM4 -> SERVO2

massigrohmann, 2022-05-12:
i had the same problem and now i solved!!... thanks a lot to everyone!!! thanks!!