iNavFlight/inav #2826: Matek F405-OSD Analog RSSI with FrSky L9R

Pairan, 2018-02-24:
I know this has been an old topic which has been fixed in 2017, many people made a video BUT this somehow doesn't work on my setup

### Board and Version
_Matek F405-OSD_
_iNav v1.8_
_INAV/MATEKF405OSD 1.8.0 Nov  1 2017 / 06:51:22 (912d1315)_

### Behavior
* _feature RSSI_ADC_ is enabled
* _FrSky L9R rssi pin is connected to the Matek rssi pin_
* _OSD indicates 29% OR 99%_ but nothing in between
* _disabling the `FEATURE RSSI_ADC` causes the rssi to be 0 all the time!_

Can anybody confirm with a MATEK FC ? As others got it running with the Omnibus, this remained on my  "need to investigate it" list :) As I failed to solve it, I'll hand it over to here

shellixyz, 2018-02-24:
Could you check if the RSSI pad is directly connected to the PB1 pin ?

Pairan, 2018-02-26:
I guess you need to help me out- what is PB1 ?

skaman82, 2018-02-26:
@Pairan 
[image]

Pairan, 2018-02-26:
Okay, after having taken a closer look it seems impossible to figure where the rssi pad is connected to. The PCB is covered with dark paint and there are so many components on its way to the chip. I tried to take a photo and zoom in ... didn't help that much though!

fiam, 2018-02-26:
@Pairan Use a multimeter in continuity mode with the FC powered off. Following the trace is next to impossible in most cases.

Pairan, 2018-02-26:
... good idea, hermano! Pero I don't have one at hand - soonest attempt would be wednesday, which kinda suxx

fiam, 2018-02-26:
No problema, amigo. Let's assume the PB1 pin is the right pin in the meantime. From the board page, it looks like the manufacturer says it supports FrSky PWM RSSI. Note that's the RSSI output from e.g. the X8R which is a PWM signal that needs to be converted into an analog voltage for the FC to read (usually via an RC filter to smooth it).

However, the L9R uses a "bare" analog voltage from 0 to 3v3 to output the RSSI which must be passed directly to an FC input to read. If you pass that through a filter designed to smooth the PWM signal, it won't work correctly.

If that's the case and you are not using the CURR pin, you can make it work by connecting the L9R RSSI to the CURR pin and swapping `rssi_adc_channel` with `current_adc_channel` via CLI.

Summoning @MATEKSYS, just in case they can clarify and/or send me an F722. That's a nice FC! 😆

shellixyz, 2018-02-26:
@Pairan It is possible the RSSI pad is connected directly to another MCU pin. If you find it it is possible you could to make a custom build using this pin instead. Otherwise you might have to modify the board to allow using an analog RSSI signal like in #2821.

Pairan, 2018-02-26:
I just checked my DIFF on V1.8 which says nothing about that setting, so I will assume it to stay on default at the moment! Is this, what you want me to try, @shellixyz :
`set rssi_adc_channel = 3` 
I will see if this changes anything later ...

shellixyz, 2018-02-26:
@Pairan That won't do anything. `rssi_adc_channel` should already be set to 3. What fiam suggested is connecting your receiver RSSI wire to the current sensor input of your board and

```
set rssi_adc_channel = 2
set current_adc_channel = 3
```

What I was suggesting is to use a multimeter to find if the RSSI pad is directly connected to any pin of the F405 MCU. But the solution fiam suggested is easier if you don't use the current sensor input pin.

Pairan, 2018-02-26:
The current pad is free as I use the FCHUB to report current readings to the board. If that wont change on changing the ADC channels, then I will sure try fiam's idea. Otherwise we'll figure out readings on Wednesday evening.

shellixyz, 2018-02-26:
But isn't the FCHUB connected to the current sensor input pin through the flat flex cable ? How would it report current readings otherwise ? If it is using the current sensor input pin on the board won't work.

Pairan, 2018-02-26:
could be ... and is pretty much likely to be! I mean, somehow the data has to flow to the right pin, right? :-)

As I'm not really into all the connections, I can only make humble assumptions. So, it looks like we'll figure out the pin on wednesday then

shellixyz, 2018-02-26:
Well it could use I2C but I don't think it is the case.

Feldsalat, 2018-02-26:
Signal from RSSI Pin can be found here:

[image]

Pairan, 2018-02-26:
Danke @Feldsalat

shellixyz, 2018-02-26:
Well that's pin 27 (PB1) if I'm not mistaken so it should work.

Pairan, 2018-02-27:
Okay ... is there anything I can do to progress here ?

shellixyz, 2018-02-27:
Feldsalat told us that the signal can be found on the right pin but what signal ? The smoothed PWM or the analog signal ? I'm still not convinced the RSSI pin is connected to PB1 directly because if you watch his picture you can see he did not do a continuity test as he connected an oscilloscope probe (PB1) and 2 feed connections RSSI + GND. That and the fact that it is not working for you Pairan it probably isn't.

Then you have 4 solutions:
1. Use another receiver with telemetry or PWM RSSI output
2. Free the current pin and use it for the analog RSSI
3. Modify your board to have a direct connection between the RSSI pad and the pin 27
4. Find a way to convert the analog signal to PWM to feed the RSSI pin (Arduino with a small program for example)

Pairan, 2018-02-27:
uh ... i think I will try to find out tomorrow - but trying to solder a direct connection to that pin will get somewhat of a job :) ...

shellixyz, 2018-02-27:
You can probably find another component connected directly to PB1 and solder to that instead. Would be easier.

Pairan, 2018-02-27:
lets hope so :D

shellixyz, 2018-02-27:
More info from the Slack channel from someone I believe to be working for Matek seems to indicate the RSSI pin should be connected to PB1 through a 1K resistor (no PWM filter).

shellixyz, 2018-02-27:
If your multimeter can measure frequency you could try checking with it if the RSSI output of your receiver is indeed analog. Otherwise add a RC filter between your receiver and the RSSI input of your FC and see if it solves the issue. Something like: L9R-RSSI ----- R10k ----- FC-RSSI ----- 10µF-cap ------ GND.

Pairan, 2018-02-27:
Some news: 
Just received the readings from @Feldsalat !
The  Matek 405 Board reports RSSI from 0-1,6V to something about 0-50%. If the voltage is raised above 1.6V then it reports about 100% almost imediately ! 

This matches to what I see while flying! Jumping between 29% and 99% assuming the signale from the L9R to be about 1.6V and above or something.

shellixyz, 2018-02-27:
The reported RSSI value should be linear from 0V (0%) to 3.3V (100%). Something is not working correctly. You could try:

```
3.3V -----  POT ------ GND
             ^--------- FC RSSI input
```

The RSSI value should match the position of the potentiometer from 0% to 100%.

Pairan, 2018-02-27:
I can try to connect a X8R (same layout) to it and see, if we get a reading as it is expected to be. If so, the L9R could be the problem.

Feldsalat, 2018-02-27:
1.6V at RSSI Pad and! also measured at pin 27 (PB1) shows 50%
at any voltage higher than 1.65V it jumps immediately to 100%

digitalentity, 2018-02-27:
If you are on 1.8 - please upgrade to 1.9-RC2

Pairan, 2018-02-27:
Check ... will upgrade to V1.9RC2  and report back in

Feldsalat, 2018-02-27:
:-) working now

Pairan, 2018-02-27:
Mine still is on 29%

Pairan, 2018-02-27:
Dirk, what have you changed that I left out !? :D

Feldsalat, 2018-02-27:
??
-V1.9RC2
-feature RSSI_ADC
nothing more

Wenn der Eingang offen ist, hast du schwankende Werte im niederen zweistelligen Bereich, sonst hätte ich jetzt keine Idee.

Pairan, 2018-02-27:
solved it!
```
#SET RSSI_SCALE
rssi_scale = 30

#set RSSI_SCALE = 100
#SAVE
```

Thanks for the support everybody!

closed 2018-02-27 by Pairan