iNavFlight/inav #2834: ANTI_WINDUP triggered at inappropriate time on fixed wing aircraft.

Daemon42, 2018-02-25:
### Board and Version
Board: CL RACINGF4 V2 
Inav Version: 1.8 but bug is older than that

### Behavior
Description: On a fixed wing aircraft, if I drop the throttle to zero (below min_check) and then change flight mode from Acro to anything else and back to Acro, I lose all Iterm stabilization.    This is a real problem on a fixed wing using the PIFF philosophy of having minimal P, and relying mostly on I and FF.   Losing Iterm means there's effectively no stabilization.  

Steps to reproduce: 
Fixed wing, with AIRMODE always enabled.   Start in Acro mode.  
1. Power up with throttle at zero, mode Acro.
2.  Arm.   Move aircraft by hand.  No Iterm response in the stabilization.   
3.  Arm, throttle up, move aircraft by hand and verify full stabilization.
4. Throttle down, verify still have full stabilization (normal Airmode behavior).    
5. Change mode from Acro to anything else, then back to Acro.   Stabilization mostly disabled, as Iterm response goes away again.  

Expected Results: 
In steps 2 and 5 above, I expect to see full stabilization the entire time Airmode is active after arming.   It's not a multirotor where it'll freak out while landing with the throttle at zero.  

* _Actual Results_
See above. 

_Is this a new feature request?_
No.   It's a bug. 

### Bug history 
I did some investigation of the code in fc_core.c where the problem comes from, and I found
this in the git log.   I believe that while trying to fix Bug #157, which had to do with issues
with losing stabilization while in automated flight modes, someone recognized that the fix
caused a side effect of disabling stabilization for fixed wing aircraft at low throttle, and they tried to fix it.
```git log -L 576,577:fc_core.c
commit 43328c2fd2a72475ebcb67890b9e4dba45900b67
Author: Konstantin Sharlaimov (DigitalEntity) <konstantin.sharlaimov@gmail.com>
Date:   Mon May 9 11:53:40 2016 +1000

    PID: Fixed I-term low authority at low throttle. Fixes #157

diff --git a/src/main/mw.c b/src/main/mw.c
--- a/src/main/mw.c
+++ b/src/main/mw.c
@@ -482,2 +482,2 @@
-                if (rollPitchStatus == CENTERED || (feature(FEATURE_MOTOR_STOP) && !STATE(FIXED_WING))) {
+                if ((rollPitchStatus == CENTERED) || (feature(FEATURE_MOTOR_STOP) && !STATE(FIXED_WING))) {
                     ENABLE_STATE(ANTI_WINDUP);
```

However, parens they added have no effect on the logic.   The underlying issue remains, which is that on fixed wing aircraft, at zero throttle (anything below min_check), if roll/pitch  are close to center, (which always happens after any mode change), there's no Iterm stabilization at all. 

### Possible fix
I'm pretty sure the intent was to do the parens like this, which ensures that ANTI__WINDUP is disabled for all fixed wing aircraft, because it's not appropriate, once the aircraft is armed.  
```
                if (((rollPitchStatus == CENTERED) || (feature(FEATURE_MOTOR_STOP))) && !STATE(FIXED_WING)) {
                    ENABLE_STATE(ANTI_WINDUP);
                }
                else {
                    DISABLE_STATE(ANTI_WINDUP);
                }
```

I tested my own target this way on a fixed wing aircraft, and it works as expected.  I have full stabilization regardless of throttle position (TPA dependent).

shellixyz, 2018-02-28:
Indeed I think there is something wrong there. But shouldn't the condition be that instead ? Enable ANTI_WINDUP if roll/pitch are centered and the feature MOTOR_STOP is enabled when not on fixed wing ?

`if ((rollPitchStatus == CENTERED) && (feature(FEATURE_MOTOR_STOP)) && !STATE(FIXED_WING))`

Daemon42, 2018-02-28:
I agree that the Fixed wing check should be up front to keep it isolated.    The other two conditions are meaningful only to multirotors, so it wasn't clear to me whether it should use ANTI_WINDUP when sticks were centered AND motor stop feature enabled,    or sticks centered OR motor stop enabled. 
I would think that with motor stop enabled.  The original logic seemed to imply either condition could trigger it, which is an OR.

shellixyz, 2018-02-28:
To me the condition as I modified it makes more sense (AND) and is now corresponding to the comment present on the previous line.

> // ANTI_WINDUP at centred stick with MOTOR_STOP is needed on MRs and not needed on FWs

Daemon42, 2018-03-01:
Sorry to drag this out, but I want to ensure the fix doesn't break multis.   

Yes your logic certainly matches the comment, but I'm not sure I trust that interpreting the "with" in the comment as an AND is best for a multi.  It's not at all clear why MOTOR_STOP was originally added to the logicl, but it is clear that adding that AND will absolutely change the behavior for most multis.  
Here's the comment a few lines above this logic
```
    /* In airmode Iterm should be prevented to grow when Low thottle and Roll + Pitch Centered.
       This is needed to prevent Iterm winding on the ground, but keep full stabilisation on 0 throttle while in air
       Low Throttle + roll and Pitch centered is assuming the copter is on the ground. Done to prevent complex air/ground detections */
```
Here's what the old logic effectively did for multirotors (ignore fixed wing, since it was clearly broken).
```
if (min throttle AND armed AND airmode on)
    if (inputs centered    OR motor_stop) 
       enable ANTI_WINDUP;
```
Following that logic down to the left side of the OR  matches the intention of the initial comment.
I think motor_stop condition was added later for reasons unknown.  

Thing is, you would almost never purposely combine Airmode on and MOTOR_STOP.  Airmode needs the motors to stay at or above idle to work effectively at zero throttle.   
Thus, if you make MOTOR_STOP a required condition in the new logic then ANTI_WINDUP will pretty much be disabled all the time.   
I'm thinking it was originally added as an OR condition just in case someone did accidentally enable AIRMODE with MOTOR_STOP  in which case you don't want iTerm winding up at all as you could have some motors completely stopped and others going fast, even while it's sitting on the ground, and you can't take off in that condition (it'll just flip). 
Thus by old logic min throttle AND armed and airmode ON and MOTOR_STOP results in ANTI_WINDUP enabled. 

FWIW, Betaflight code used to do this. 
```
    /* In airmode Iterm should be prevented to grow when Low thottle and Roll + Pitch Centered.
     This is needed to prevent Iterm winding on the ground, but keep full stabilisation on 0 throttle while in air */

-    if (throttleStatus == THROTTLE_LOW) {
-        if (IS_RC_MODE_ACTIVE(BOXAIRMODE) && !failsafeIsActive() && ARMING_FLAG(ARMED)) {
-            if (rollPitchStatus == CENTERED) {
-                antiWindupProtection = true;
-            } else {
-                antiWindupProtection = false;
-            }
-        } else {
-            if (IS_RC_MODE_ACTIVE(BOXAIRMODE)) {
-                pidResetErrorGyroState(RESET_ITERM);
-            } else {
-                pidResetErrorGyroState(RESET_ITERM_AND_REDUCE_PID);
-            }
-        }
```
Notice the exact same initial comment.    No reference to MOTOR_STOP in the code at all.
And then all that logic was removed in 2016 and replaced with
```
    if (throttleStatus == THROTTLE_LOW && !airmodeIsActivated) {
        pidResetITerm();
```
while the same comment remains, which doesn't match the code at all, which is why I don't necessarily trust comments.   Iterm windup is now only prevented when throttle is low and airmode is Off.    BF no longer cares about centered inputs and it never cared about MOTOR_STOP.    There may be some additional trickery deeper in the stabilization loop to prevent the airmode bounce, or they just decided most folks disarm to land now.  

It's very clear that we don't want ANTI_WINDUP for fixed wings.   That's easy enough.
I'm just afraid that putting that AND in there will break the expected behavior for multis, and bring the airmode landing bounce back.     The question is, does it make more sense to have the OR MOTOR_STOP there to handle a weird edge condition, or just remove it entirely?

digitalentity, 2018-03-01:
I agree with @Daemon42. We need to make sure that anti-windup behavior stays **exactly the same** for multicopters. Thus I suggest adding a one top-level condition that checks for `FIXED_WING`:

```
if (STATE(FIXED_WING)) {
    // fixed-wing anti-windup code goes here (essentially only check for airmode)
}
else {
    // multirotor anti-windup code goes here (current code with OR condition)
}
```

Daemon42, 2018-03-01:
Even with airmode, there's no need for any anti-windup code for fixed wing.    If you want it more clear could change the existing logic from 
```
if ((rollPitchStatus == CENTERED) || (feature(FEATURE_MOTOR_STOP) && !STATE(FIXED_WING))) {
                    ENABLE_STATE(ANTI_WINDUP);
                }
                else {
                    DISABLE_STATE(ANTI_WINDUP);
                }
```
to
```
if ( !STATE(FIXED_WING) {  // multirotor only
    if ((rollPitchStatus == CENTERED) || (feature(FEATURE_MOTOR_STOP)) {
                    ENABLE_STATE(ANTI_WINDUP);
                }
                else {
                    DISABLE_STATE(ANTI_WINDUP);
                }
   }  // no else
```

digitalentity, 2018-03-01:
@Daemon42 most small airplanes do perfectly fine without anti-windup, but it's incredibly hard to launch a large and heavy flying wing on a bungee if wind-up at zero speed is not accounted for. Originally I've tried removing anti-windup for fixed wings alltogether and ended up glueing more than flying.

Anti-windup measures at zero throttle are a must, but conditions to allow them must be chosen very carefully.

digitalentity, 2018-03-01:
This is what I have in mind:

```
static void handlePIDAntiWindup(throttleStatus_e throttleStatus)
{
    static bool antiWindupWasActivatedOnce = false;

    // Track if ANTI_WINDUP was activated
    if (!ARMING_FLAG(ARMED)) {
        antiWindupWasActivatedOnce = false;
    }

    if (FLIGHT_MODE(MANUAL_MODE) || !ARMING_FLAG(ARMED)) {
        DISABLE_STATE(ANTI_WINDUP);
        pidResetErrorAccumulators();
        return;
    }

    // At non-zero throttle - always disable ANTI_WINDUP
    if (throttleStatus != THROTTLE_LOW) {
        DISABLE_STATE(ANTI_WINDUP);
        return;
    }

    rollPitchStatus_e rollPitchStatus = calculateRollPitchCenterStatus();

    if (STATE(FIXED_WING)) {
        if (isAirmodeActive()) {
            // On airplanes only activate ANTI_WINDUP if throttle = low, roll/pitch = center and ANTI_WINDUP was never activated this flight
            if ((rollPitchStatus == CENTERED) && !antiWindupWasActivatedOnce) {
                ENABLE_STATE(ANTI_WINDUP);
                antiWindupWasActivatedOnce = true;
            }
            else {
                DISABLE_STATE(ANTI_WINDUP);
            }
        }
        else {
            DISABLE_STATE(ANTI_WINDUP);
        }
    }
    else {  // MULTI_ROTOR
        if (isAirmodeActive() && !failsafeIsActive()) {
            if ((rollPitchStatus == CENTERED) || feature(FEATURE_MOTOR_STOP)) {
                ENABLE_STATE(ANTI_WINDUP);
            }
            else {
                DISABLE_STATE(ANTI_WINDUP);
            }
        }
        else {
            DISABLE_STATE(ANTI_WINDUP);
        }
    }
}
```

digitalentity, 2018-03-01:
For `antiWindupWasActivatedOnce` to be set to `true` you have to have throttle above zero and roll/pitch not at center **at the same time**. I doubt you wiggle the ailerons with motor running when you test. Also, you can wiggle the sticks when disarmed - servos will still move. 

There's no point in comparing INAV to other stabilizers. Users launch it in ANGLE mode using dead simple 4-channel radios which can't do any mode switching. If we can reduce the possibility of throwing the plane with control surfaces saturated before plane is in the air - we should do it.

Daemon42, 2018-03-01:
Ok.  I guess I don't really care what happens in the first 2 seconds of the flight, as long as this loss of
stabilization can't be easily/randomly triggered again in the air.  Relying on roll/pitch at center was never enough since that resets every time you switch flight modes.  Because I like to soar and I'm used to various other airplane gyros and stabilizers, I noticed the loss of stabilization almost immediately the first time I flew iNav when I switched from Passthru to Acro while checking trim.  Suddenly the plane wouldn't track straight.   Later after I'd investigated the issue, I mentioned what I had found in a  thread with a bunch of other fixed wing iNav users, and a few folks chimed in with some variation of 
"Oh, I noticed loss of stabilization while gliding too, but just thought there must be some good reason for it so I never said anything."   

Whatever we can do to get this fixed, would be much appreciated.

digitalentity, 2018-03-01:
What I suggest keeps the current logic of enabling the anti-winfup, but limiting that for once per arming. So unless you disarm in air you won't get any loss of stabilization. We'll target this to 1.9.1 - I don't want to delay 1.9 any further.

PS: It just came to me that if your plane requires I-term to successfully stabilize it might be either not trimmed or not tuned well (not in all cases, but in some - definitely).

Daemon42, 2018-03-01:
Ya, we can work with your fix. 

Re PS.. 
The whole point of using PIFF for fixed wing is that most of the stabilization comes from the Iterm, 
FF handles user input (fairly direct from stick to surface), and a little bit of P is there only to sharpen the responses.  This is spelled out the iNav PIFF tuning guide. 
If you let P do most of the work and tune for cruise speed, then you'll get speed wobbles at higher airspeeds.  
TPA can help alleviate that, but only if you're using the throttle, which in many cases I'm not (which is why I noticed this issue in the first place).    I'm slope/thermal soaring with the throttle at zero, but I can still be flying quite fast in a dive.  Since there's no other way to scale the PIFF response on the fly (no gain knob equiv), I have to keep P low, and use Iterm for most of the stabilization, per PIFF tuning recommendation.

Daemon42, 2018-08-27:
Just a friendly reminder this is still out there.    
Also I notice the Enhancement and Feature request tags were put on this bug. 
It's not an enhancement.  It really is a bug.   I-term is the primary stabilization mechanism in Acro mode especially in really turbulent air (you need the aircraft to return to its original orientation on it's own). 
Currently if you chop the throttle (say you want to glide in for landing), you lose most of your stabilization.   If you want to soar with the motor off, you lose most of your stabilization.     I do not believe that was ever the original intent of the code change that created this bug (per the discussion above).

Daemon42, 2019-01-10:
I saw above that the fix was targeted to 2.1 so I've been waiting, but can see that it still hasn't been addressed in 2.1.0_RC1.  
It's been almost a year since I report this and yet the most effective part of the stabilization loop is still disabled, in the air, any time one switches modes while throttle is at zero (gliding).    This makes for very bumpy soaring,  and crappy power off landings.

digitalentity, 2019-01-11:
Nobody submitted a PR for it

fiam, 2019-01-29:
We should postpone this until we implement #2331, it would be the best solution for this problem.

Daemon42, 2019-01-30:
I disagree.  I think trying to detect when you're flying, versus sitting, versus launching, is a *much* more challenging problem to solve, and waiting till you figure that out just means this issue gets put off forever.    For those of us who soar, or want to land with the motor off, this issue represents a fundamental flaw in the core stabilization logic of iNav.     In a PIFF loop, the I-term does most of the work, and it's completely disabled here, in the air.

DzikuVx, 2021-08-23:
solved by #6076

closed 2021-08-23 by DzikuVx