iNavFlight/inav #2836: POS HOLD Mode Intervention - change position instead manual override

b14ckyy, 2018-02-25:
As soon as i know during my last drone flights with 1.7.3, when drone is in POS HOLD mode and I input on pitch/yaw, it temporary disables POS HOLD and goes back into ALT HOLD + ANGLE only. 

That works fine in calm weather and no wind, but if it is a bit windy and i want to do just slight position corrections, the drone will instantly drift with the wind until I give enough input to fight against it. 

Also when there is wind from side and I just want to move forward or flying forward in general, the drone drifts away sideways. 

I'd recommend not to control in angle mode on stick input but instead, control just a calculated GPS position. 
Position change speed should then be cruise speed on max stick deflection. So you could still fly a straight line because the quad is compensating all wind from itself and even drifting should be compensated. 

As a safety function, the copter should instantly stop and hold position again if sticks are centered, instead of continuing to calculated GPS position.

DzikuVx, 2018-02-25:
The thing you are looking for is CRUISE. Look in Configurator in Advanced Tuning

closed 2018-02-25 by DzikuVx

b14ckyy, 2018-02-25:
Oh damn sorry. Overlooked that in all the wiki pages. A quicktip in configurator would be nice here :)