iNavFlight/inav #2840: Holybro Micro M8N GPS Module Wont Hold FIX - iNav setting?

Jason51873, 2018-02-26:
I have now tried 2 different brand new Holybro Micro M8N GPS Modules and they both do the same thing that's why I think this must be a setting or issue with iNav - please see video:

https://www.youtube.com/watch?v=pAT2jCwO3b4

Thieu-s, 2018-03-16:
Any news yet?

I have exactly the same problem as in the video above. On 2 different crafts, both with an Omnibus F4 Pro (one original and one clone) and 2 different M8N GPS modules from HobbyKing. Tested on different firmware 1.8 and 1.9. I've tested litterly all posible settings for GPS... All the same...

In the upper left corner the osd shows the blinking sats...

https://www.youtube.com/watch?v=pl3PWt3aqgE

shellixyz, 2018-03-18:
The issue seems to be related to the Holybro Micro M8N GPS. Other M8N module I tested are working fine with the Omnibus F4 PRO and other FCs. Unfortunately I don't think any active developer have access to this GPS module for testing.

P-I-Engineer, 2018-03-29:
i'm having the same issue with the bn-220 gps https://www.banggood.com/Dual-BN-220-NEO-M8N-GPS-GLONASS-Antenna-Module-Ublox-M8030-TTL-Level-p-1208588.html?rmmds=search&cur_warehouse=CN

i've tried with a omnibusf4 pro and a matek 405 mini. I previously worked on a omnibus f3 pro. I no longer have the F3 to confirm.

jelle737, 2018-04-07:
I have 2 holibro gps': 1 working good a long time and 1 yet to maiden, I had a similar problem that the packet error went up every few seconds. Through µ-center using gpspassthrough I connected up with the gps and compared the settings of the good one with the 'broken' one.
What I saw is that in the 'broken' one had next the the regular 4 UBX - NAV messages (that are POSLLH SOL, STATUS and VELNED. Also a few other messages reporting sporadic.
I turned of all the messages that were reporting once in a while or continiously that were not one of the 4 above.
The second thing I found is that the configuration page UBX - CFG - INF had everything checked there, I unchecked everything. Unsure what that is but it's kind of a selection of type of messages allowed to pass?

After getting the settings in the 'broken' one identical to the 'working' one, everything seems fine now. I suspect the same should happen on a resetup of the gps, by turning gps off in iNav config and turning it back on after reboot, since iNav does setup the GPS right? Anyway to me it felt very much settings related and since I messed around with µ-center before easy to check and fix.

I can share how the 'good' settings looked like (took a few printscreens in the progress) if someone has a use or like to compare to fix his manually.

evmoius, 2018-04-13:
@jelle737 could you elaborate on this?  I have the same problem. 2 Holybro Micro M8N GPS units. One works great, the other won't hold a fix. I'd like to try your solution but I'm not quite sure what you did.
I'm using a Matek F405 AIO.

jelle737, 2018-04-13:
A question to the developers: Is there a way to force the RC to re-setup the GPS? I thought setting up the GPS happens automatically but somehow it does not for those above that have problems.

@evmoius actually it is described how to manually setup a GPS unit here: https://github.com/cleanflight/cleanflight/blob/master/docs/Gps.md#ublox-gps-manual-configuration
You need u-center (v8.25) software, through iNav configurator in the cli write 'gpspassthrough', then disconnect the configurator software and connect with u-center at a baud rate of 38400 (or the one you set in the configurator ports tab). An easy way to activate/disactivate messages is double clicking them. When messages are send their name becomes black instead of grey, that way you can find the one that are transmitting that should not transmit. Don't forget at the end you have to save current configuration and send that to confirm.

Good luck. I could not find any visual (youtube alike) video that shows the progress so words will have to do for the moment.

spak01, 2018-04-15:
@jelle737 that gps configuration guide you linked is either incorrect or not updated.  I went searching through the code to look up what all the autoconfig was doing.  There is quite a discrepency between that old wiki everyone keeps linking vs what the code actually is doing.
First few things I noticed off the bat.....if you enable Galileo the autoconfig drops the rate to 5hz.
Also the auto config changes the uart 1 protocols to some thing different than what every gps wiki mentions.  It does not use the stated 0+1+2 and the 0+1... autoconfig will config it as 0+1 and 0 respectively.
Also look below for the difference between UBLOX and UBLOX7...
Some of these msgs being enabled and disabled have never been specified in any wiki or other documentation.

I am just asking as since I have finally been trying to use 1.9.0....the gps setup has been giving me a lot of trouble.
[image]

Either we are missing documentation or the code is doing something we never knew about.

thedarknebula, 2018-04-17:
@evmoius How did you perform the reset in U-center? I am having signal dropping issues discussed in this thread but want to be careful to not cause further problems in my setup.

P-I-Engineer, 2018-04-17:
my problem was because my gps died, don't know why. I've replaced with the HGLRC m8n gps, so far, in bench testing, everything appears to be working.

evmoius, 2018-04-17:
@thedarknebula Open the configuration window and select CFG.  Then select ""Revert to default configuration" and click "Send"

shaiss, 2018-05-01:
@evmoius cfg->cfg->revert to defaults.  its buried.  GPS resets to 9600 baud rate, so remember to change that in inav.

closed 2018-10-23 by stale[bot]