iNavFlight/inav #2843: Adding an AUTO ON/OFF FAA Altitude Height limit of 400FT option.

alanfj, 2018-02-27:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

shellixyz, 2018-03-06:
Linked to #1510 and #2322

giacomo892, 2018-03-07:
@alanfj for the moment you can set an altitude alarm on the OSD. No one will ever come and say you were breaking the altitude limit by few meters...

closed 2018-05-27 by stale[bot]