iNavFlight/inav #2848: Add temperature to OSD & blackbox

digitalentity, 2018-02-28:
Dependent on https://github.com/iNavFlight/inav/pull/2846

jelle737, 2018-03-09:
Hi, I'd like to see temperature implemented to OSD and got directions on how to approach this. Maybe you can set me further in the right directions to solve this correctly. I know from the dependent above the plan is the have the accelerometer report the temperature, and my approach is to have the barometer report the temperature.

Actually betaflight has code about temperature coming from the core
https://github.com/betaflight/betaflight/blob/f97df6827e968a827e05e5be60fa80194d308e60/src/main/io/osd.c#L768-L772

The solution there would be:
```
#ifdef USE_BARO
    case OSD_CORE_TEMPERATURE:
        if (sensors(SENSOR_BARO)) {
            tfp_sprintf(buff, "%3d%c", osdConvertTemperatureToSelectedUnit((int16_t) (baro.baroTemperature / 100)), osdGetTemperatureSymbolForSelectedUnit());
        }
        break;
#endif
```

But since it's not betaflight and I'd like to see temperature added rather then replacing another feature, just under this line, the end of `#if defined(USE_BARO) || defined(USE_GPS)`
https://github.com/iNavFlight/inav/blob/dc4ef594fa7c43690301e8f2aa2602c582b79f49/src/main/io/osd.c#L1261

I'd like to add:
```
#if defined(USE_BARO)
    case OSD_TEMPERATURE:
        {
            tfp_sprintf(buff, "%5df", (baro.baroTemperature + 50)/ 100); //Airmamaf
            buff[5] = SYM_TEMP_C;
            break;
        }
#endif
```
or a variant that is more solid and allows for changing between `°C` and `°F`
This also needs an `#include "sensors/barometer.h"`

But I believe it might not be as easy as I might hope, as the issue states there is also blackbox, and the configurator for toggling and positioning this, so I need some guidance.

Credits: @efendy

digitalentity, 2018-03-09:
@jelle737 would be better to use `getTemperature()` API once https://github.com/iNavFlight/inav/pull/2846 is merged.

jelle737, 2018-03-09:
Interesting, that's why I thought first to ask, as I saw there was already some development in this area.

danarrib, 2018-07-05:
I don't know if my opinion worth something... But I vote to see this feature implemented. The guy who made the famous video of 10km altitude did it... Maybe he can kindly make a PR of his work.

jelle737, 2018-07-07:
@danarrib That is where my comment above is based upon. They did the trick by replacing a data field (/hacking it in the code) with the temperature instead of nicely implementing it.

putimir, 2018-08-15:
++ here. Has anybody implemented it? Can you share link to the code?

timman2er, 2018-09-30:
@jelle737 for reference, betaflight has a function already for selecting the symbol, see https://github.com/betaflight/betaflight/pull/6656/files#diff-8d20d724db9c8b628f9799a04fab9dcaR1527 .

digitalentity, 2019-01-06:
Fixed

closed 2019-01-06 by digitalentity