iNavFlight/inav #2849: Omnibus F4 V3 flight controller” HORIZON along with ALTHOLD stabilization wouldn't respond.

felix44860, 2018-02-28:
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here:

Omnibus F4 flight controller mounted in a “S670 Swallow fixed wing.”
Size: 36x36mm
Mount holes: 30.5x30.5mm
STM32 F405 MCU, Runs Betaflight 3.0.1 firmware, could run faster loop time. 
Using F4 MCU controls OSD over SPI bus in DMA mode.
OSD Configation has been included in Betaflight GUI. 
SBUS/PPM input (Pinheaders)
6PWM output (1-4Pinheaders and Sh1.0 Plug, 5-6 as Pinheaders)
With BARO
SPI Sensor MPU6000
Included 128Mbit Flash, please always use BEC for the FC.
Supported by BETAFLIGHT and dRonin

_Use CLI command `version` and put its output here_

Link to the S670 Swallow Omnibus F4v3 flight controller CLI: (https://pastebin.com/XzEyVD3w)

* _Describe the problem_
Servos are connected to a 5v external source.
When I engage the “HORIZON” modes along with “NAV ALHOLD”
* _Steps needed to reproduce the problem_
Once I engage the modes and I hold the wing on my hands and move it around. The wing ailerons do not try to stabilized.
* _Expected Results_
I am using the same type Omnibus f4 v3 flight controller in another wing and when I engage the “HORIZON” modes along with “NAV ALHOLD” the ailerons try the stabilized the wing.

shellixyz, 2018-02-28:
Please try to upgrade and see if it fixes your issue. 1.7 is old.

felix44860, 2018-02-28:
Okay, thx!

Sent from Felix's iPhone

felix44860, 2018-02-28:
Thank you, here is a youtube video I just posted :
https://youtu.be/lsqIHMVGuNo

Thank you for your help!

digitalentity, 2018-03-01:
This is a known bug. Fixed by https://github.com/iNavFlight/inav/pull/2060

closed 2018-03-01 by digitalentity

felix44860, 2018-03-01:
Thank you!

Sent from Felix's iPhone