iNavFlight/inav #2854: Full battery is in WARNING status

digitalentity, 2018-03-01:
```
# status
System Uptime: 24 seconds
Current Time: 0000-01-01T00:00:00.000+00:00
Voltage: 12.51V (3S battery - WARNING)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: None inserted
Stack size: 6144, Stack address: 0x10010000
I2C Errors: 0, config size: 3390, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 2000, PID rate: 500, RX rate: 9708, System rate: 9
Arming disabled flags: NAV RX THR CLI

# set vbat
vbat_adc_channel = 2
Allowed range: 0 - 4

vbat_scale = 1090
Allowed range: 0 - 65535

vbat_max_cell_voltage = 424
Allowed range: 100 - 500

vbat_min_cell_voltage = 330
Allowed range: 100 - 500

vbat_warning_cell_voltage = 350
```

This seems incorrect isn't it?
`Voltage: 12.51V (3S battery - WARNING)` 

@shellixyz ?

digitalentity, 2018-03-01:
Changing `vbat_max_cell_voltage = 430` fixes the issue. Must be a bug in battery monitoring code.

digitalentity, 2018-03-01:
Setting battery capacity to zero seems to fix the issue. Bug is probably related to #2855

shellixyz, 2018-03-01:
Yes it is related. The warning state is not because of the voltage but because of the capacity settings as you found out. When capacity thresholds are used the battery state is set according to them.

digitalentity, 2018-03-01:
Closing as this is not a bug, but a configuration issue.

closed 2018-03-01 by digitalentity