iNavFlight/inav #2855: Battery capacity shown incorrectly

digitalentity, 2018-03-01:
Battery remaining capacity incorrect:
[image]

```
battery_capacity = 2200
battery_capacity_warning = 1500
battery_capacity_critical = 2000
battery_capacity_unit = MAH
```

Pairan, 2018-03-01:
I was wondering about that the evening I flashed mine to RC2 ...

shellixyz, 2018-03-01:
I can't see anything wrong. You have set the critical capacity to 2000 so between full (2200) and empty (2000) you have 200mAh. Maybe you were expecting the empty left capacity in the battery to be 200 (battery empty after 2000mAh drawn) but that's not how I implemented it. It is working like the battery voltage. `battery_capacity_critical_capacity` is the equivalent of `vbat_min_cell_voltage` but for the capacity. To set `battery_capacity_critical` I generally take 20% of the battery capacity. I find it pretty convenient. For a 2200mAh, `battery_capacity_critical` = 2200*0.2 = 440. Meaning it is considered empty when 20% of the capacity is left.

shellixyz, 2018-03-01:
So the right settings for what you wanted to do are:
`battery_capacity = 2200`
`battery_capacity_warning = 700`
`battery_capacity_critical = 200`

digitalentity, 2018-03-01:
Then we need to change the names to be less confusing. Currently it could be easily confused like this:
Capacity = total mAh which could be taken from battery
Warning capacity = amount of mAh we can use before hearing the "warning" beeper
Critical capacity = amount of mAh we can use before hearing the "battery critical" beeper
Now I understand that it's wrong, but that's what my initial understanding was.

Battery capacity alarm in OSD tab has the same meaning - amount of capacity we use before triggering the alarm.

Suggestion: change from "capacity" to "percentage". This can be done at least at UI level (Configurator). It's easier for people to understand "percentage".

digitalentity, 2018-03-01:
Closing, not a firmware issue :)
Docs are amended to explain the new settings (#2856). Probably nobody will read them, but Configurator is going to be more user-friendly as well. 
Thanks @shellixyz

closed 2018-03-01 by digitalentity