iNavFlight/inav #2859: I-BUS telemetry on cc3d board

DaniilDem, 2018-03-01:
I have a problem with telemetry. I am use inav 1.7.3. with cc3d. I connected i-bus sens to tx flexi port (main port too) and enabled telemetry and set ibus telemetry on port. But this don't work.
[Here](https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20CC3D.md) writed cc3d only LTM telemetry it true? Can i will compile inav1.7.3 with ibus?

shellixyz, 2018-03-01:
IBUS telemetry is not compiled in for the CC3D target in release builds. You can build a custom version with it enabled. I can't guarantee it will work but probably. Be aware that support for flight controllers with the STM32F1 MCU is being dropped for later versions. Version 1.7.3 is the last one with a release build for the CC3D FC.

Here is a hex file for the CC3D with IBUS telemetry enabled:
[inav_1.7.3_CC3D.hex.zip](https://github.com/iNavFlight/inav/files/1772010/inav_1.7.3_CC3D.hex.zip)

DaniilDem, 2018-03-01:
Thank! it really work firmware.

closed 2018-03-31 by DzikuVx