iNavFlight/inav #2861: Ability to command a dive in a mission

AAllport, 2018-03-01:
Question, 
I am working on a project that will make use of a delta wing making a freefall dive and then pull out and land at a predefined location & height. Is it currently possible to do this in inav?

shellixyz, 2018-03-05:
You can make the wing dive to a waypoint by setting the altitude of the waypoint but you can't land automatically elsewhere than home for the moment.

AAllport, 2018-03-07:
Do I need to make any config changes so that it does an almost straight down dive?

shellixyz, 2018-03-07:
Yes I think you will need to set `nav_auto_climb_rate` to the vertical speed when diving and also `max_angle_inclination_pit` and `nav_fw_dive_angle` to the angle you want it to dive at.

stronnag, 2018-04-04:
This support question appears answered to the limit of iNav capability. Closing

closed 2018-04-04 by stronnag