iNavFlight/inav #2864: VER: NAV/NAZ 1.7.3 Beeper does not work

itswally1, 2018-03-02:
The beeper will not work. In the mode selection the beeper is enabled and when the proper switch is positioned it lines up under the beeper selection.
The board is EMAX SKYLINE32
The CLI says beeper disabled:None
Also on the CLI if i type
 beeper -on
The response is:
Disabled ON_USB
The beeper is assighed channel 6 and the output on the beeper pins goes from .9 volts to .3 volts

How can I enable the beeper?

Thanks in advance for your support

shellixyz, 2018-03-02:
You don't need to enable the beeper it is enabled by default. It should sound when the `BEEPER` flight mode is enabled. You should should check your wiring and beeper polarity and health.

itswally1, 2018-03-04:
Thanks for the quick response.
The board/beeper works with Cleanflight not with INAV.
INAV has better stability so that is why I am using it. 
In my note I have read the voltage at the beeper and beeper unit works with 5v applied.
The flight mode has been set prpperly

So why when I type beeper and the flight mode is on for the beeper
does the screen respond with:
 beeper disabled:None
and
Disabled ON_USB
???????
Regards,
Don

shellixyz, 2018-03-04:
Disabled ON_USB probably means it is disabled when you connect the board with an USB cable.

Does your board beep when you plug the battery ? It should beep at startup. Fast beeps then 3 slow beeps after the run-time calibration.
What target are you using with Cleanflight ?

Please post a dump of your iNav settings.

itswally1, 2018-03-15:
No beeps. Currently using INAV
Here is the dump
Thanks in advance for looking at the dump
============================
# version
# INAV/NAZE 1.7.3 Sep  9 2017 / 09:59:03 (974440df)

# dump
# INAV/NAZE 1.7.3 Sep  9 2017 / 09:59:03 (974440df)
mixer QUADX

mmix reset

servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 
smix reset

feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature PWM_OUTPUT_ENABLE
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACTION_SUCCESS
beeper ACTION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB
[... 269 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

shellixyz, 2018-03-18:
I can't try to reproduce since I don't have a Naze FC anymore. All I know was that it was working when I was using one. Sorry but the boards using F1 MCU like the Naze are not supported anymore you should consider upgrading to a F3, F4 or F7 board.

stronnag, 2018-03-18:
works fine on my bench test naze32 (Flip32 h/w) with 1.7.3.

SPlissken, 2018-03-18:
I got the same problem moving from 1.7.2 to 1.7.3:
Beeper not sounding on NAZE 1.7.3 #2252

I had a poke around in the code but I couldn't immediately see the issue.

shellixyz, 2018-03-18:
@SPlissken It's odd nothing changed between 1.7.2 and 1.7.3 related to beeper handling in the general code or the NAZE target.

SPlissken, 2018-03-18:
@shellixyz definitely odd, I went a little crazy chasing it at the time: ghosts in the interrupts.

It was really easy to reproduce on the bench: erase and flash naked Naze to 1.7.2 beeping, erase and flash naked Naze to 1.7.3 no beeping.

By naked I mean just USB, 5V and beeper connected.

itswally1, 2018-03-27:
I changed my four quads back to 1,7,2 and the beeper works. So the issue is with 1.7.3 code keeping the beeper from working.

Thanks all for you support and hopefully the issue with 1.7.3 will be fixed in the future

virtualfish, 2018-08-16:
Hello, I know that this issue is closed but I thought I could help someone with same problem.
I own a Naze32 board and I had the beeper not working with INAV 1.7.3 (974440d).
I solved it rebuilding the 1.7.3 version after having removed all the code related to the cpu_underclock.  This code is introduced after the version with commit no. 782b482 on Sept. 8 2018.
An easy way to do it is to copy the follwing files from the 782b482 version and replace them with the 974440d ones, recompile it and that's it! The beeper should be working again.

Theese are the files:
src/main/drivers/system.h
src/main/drivers/system_stm32f10x.c
src/main/fc/config.c
src/main/fc/config.h
src/main/fc/fc_init.c
src/main/fc/settings.yaml
src/main/target/common.h
src/main/vcp/hw_config.c

The underclock feature is ony for F3 and F4 processor anyway.
I did this way just because I wanted to keep all the other changes.
Kindly let me know if I did something bad or if underclocking is useful for the Naze32. 
Greetings! 
:-)