iNavFlight/inav #2871: INAV FW 1.8.1 GPS no Lock

BadgerG, 2018-03-04:
I have an SP Racing F3, most likely a clone. I have been fighting a GPS issue for a couple of weeks. I have tried 3 different SP Racing F3 boards and 3 different GPS units with similar results. Here is my  setup:

SP Racing F3 FC
INAV 1.8.1 
FRSky  X8R receiver (SBUS connection, UART 3)
HolyBro M8N GPS module with compass (UART0)
3DR Telemetry radio (915 MHz, UART2)
FRSky X9D Transmitter
King Kong 260 Quad frame

I can plug the battery in and it takes 3 to 4 minutes with the quad by a window and the GPS modules blue lock light starts flashing. The INAV configurator (1.9) shows no indication of GPS lock (GPS icon is red). Then the quad will sound a lock tone and it will have GPS with anywhere from 7 to 12 sats.

Then a few seconds later it loses lock and the icon turns red again. The GPS still show a blinking blue light. Then every 1 or 2 minutes it will get lock for a few more second 10-15 seconds at the most. Cycling power on the quad fixes nothing. I can at that point take it outside and I get the exact same result accept it will show more satellites are locked. At times I can see it going from lock to no lock and back repeatedly even though the GPS module show a blue locked light.

I can take the same GPS unit and connect it to my Tarot 680 Pro with a Pixhawk and within 15 seconds it shows lock with 11+ satellites and never loses lock. It normally shows 13 to 18 satellites locked on the original GPS unit and this was able to lock on to 16 in the house. The other GPS modules act the same way on the SPR F3 and get significantly fast lock and sat number on the Pixhawk. So in this case this is not a problem with the GPS modules or the SPR F3.

I can arm the quad and fly it after turning off the extra arming checks, and it flies pretty well. Alt hold works but need tuning.

So to me in mho, is that the GPS code is not getting or using the information the GPS unit is sending or the configuration that is sent to the GPS is not the same as what the Pixhawk sends to the GPS module. So far I have never been able to get any GPS units to work anywhere near as good under INAV and APM.

Because of the way my ports are setup I have to disable the GPS and Mag in order to connect to the USB to get blackbox logs or settings dump, so I don’t have them. 

One other thing is that I am using and external Mag on the GPS module and the mag appears to work just fine, with no I2C errors. I cut the trace to the internal mag. I did a compass calibration and it was successful. The Quad show good health except for the GPS not being locked. If this was just one FC board and 1 GPS unit I would say faulty hardware, but these are brand new FC and GPS modules.

At first I thought maybe the UART port had issues, but it works just fine for installing firmware to the boards at 115K baud so it is working fine. I really don’t want to point the finger at the firmware, but I don’t know what else could be causing perfectly good hardware not to work that works fine using different firmware on a different board.

This happens in INAV 1.9 and Mission planner for INAV.

shellixyz, 2018-03-04:
Please post a dump of your settings

BadgerG, 2018-03-04:
Okay I pull the quad apart so I can connect to the USB port to down load the dump.

# version
# INAV/SPRACINGF3 1.7.0 May  3 2017 / 11:12:57 (104b66c)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper -HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
[... 399 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 50
set pitch_rate = 60
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

shellixyz, 2018-03-04:
In your summary your were saying you are using 1.8.1 but it doesn't exist, only 1.8(.0). Then at the end of your first comment you are saying it is happening in iNav 1.9. But in your second comment it is written in your config dump that you are using version 1.7.0. I'm confused.

In your config dump you have no serial port configured with the GPS function. Wrong dump ?

BadgerG, 2018-03-05:
I install 1.7 to see if the problem happens in a lower version. I had the same issue with 1.8.

BadgerG, 2018-03-05:
I use the same port for GPS, so I have to unplug the GPS and MAG in order to down load using the USB. Otherwise it takes hours to get the data downloaded.

BadgerG, 2018-03-05:
I explained all the in my first post.

BadgerG, 2018-03-05:
Here is a video of what is happening. It is a Youtube video.

https://youtu.be/n0ytpiiqDig

shellixyz, 2018-03-05:
Please post a config dump where you have the GPS configured. It is not a good idea to have the GPS on UART1. What are you using the other UARTs for ? My guess is that it is what is causing your issue.

BadgerG, 2018-03-05:
I used the 3DR radio to get this dump so it tends to loose characters, so that is way I use the UBS to down load.

# version
# INAV/SPRACINGF3 1.7.0 May  3 2017 / 11:12:57 (104b66c)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper -HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
[... 260 lines cut ...]
set fw_ff_yaw = 60
set fw_p_levenation_rll = 300
set max_angle_inclination_pit = 300
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set dterm_setpoint_m_notch_hz = 0
set dterm_notch_cutoff = 1
set pidsum_limit = 500
set yaw_p_limit = 300itch = 0
set rate_accel_limit_yaw = 10000
set heading_hold_rate_limit = 90
set nav_mc_pos_z_p = 50
set nav_mc_
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_ve50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fwset pitch_rate = 60
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

BadgerG, 2018-03-05:
I changed the port setup so that the GPS is using UART2 and the problem is still there. 3D lock with 18 sats and it still keeps loosing lock every 3-4 seconds, then it comes back almost immediately.

shellixyz, 2018-03-05:
Maybe its caused by interference from your 3DR modem. Did you try without it ?

BadgerG, 2018-03-05:
Using UART2 does work better, it doesn't totally lose GPS, but the error count on the GPS page is high. 1744 error out of 25000 messages.

BadgerG, 2018-03-05:
Then there is no way to see if the GPS has a good lock without the 3DR radio, so no there is no way to do that. Can the USB be used while a battery is connected?

shellixyz, 2018-03-05:
Yes if if you have the GPS on UART2 you can use the battery while connected with USB

BadgerG, 2018-03-05:
Okay I connected without the 3DR radio and the results are the same.

shellixyz, 2018-03-05:
How is your GPS powered ? Are you sure it's supply voltage is the right voltage and clean ?

BadgerG, 2018-03-05:
The power is coming from the FC. I don't think the power is a problem. The UBEC supplies 1.5 amps to the FC, reciever, and one 4 led light.

shellixyz, 2018-03-05:
Well I don't know what the issue could be. I never had any issue making M8N GPS modules communicate with iNav.

digitalentity, 2018-03-05:
@BadgerG since the issue is opened for version 1.8.1 please provide a relevant dump.
Try also connecting your GPS unit to UART2 and configure accordingly.

BadgerG, 2018-03-05:
The FW version is actual 1.7.  The GPS is already connected to UART2 and I have already posted a dump. The problem is not as bad using port 2. Lock is achieved more quickly. However it still looses lock every few second for about 1/2 second repeatedly.

shellixyz, 2018-03-06:
Looks like it is linked to the HolyBro M8N GPS module. Another user has the same issue see #2840.

Thieu-s, 2018-03-17:
I have exactly the same problem as in the video above. On 2 different crafts, both with an Omnibus F4 Pro (one original and one clone) and 2 different M8N GPS modules from HobbyKing. Tested on different firmware 1.8 and 1.9. I've tested litterly all posible settings for GPS... All the same...
Even with nothing else connected, just the fc with the gps, powered by usb or lipo, same result... 

In the upper left corner the osd shows the blinking sats...

https://www.youtube.com/watch?v=pl3PWt3aqgE

jelle737, 2018-04-07:
@Thieu-s @BadgerG I commented https://github.com/iNavFlight/inav/issues/2840 with my findings, could that help you out solving the problem?

closed 2018-06-20 by stale[bot]