iNavFlight/inav #2872: MATEK F772 STD - NO OUTPUT ON S6 PAD.

flyfisher2, 2018-03-05:
There is no output on the S6 pad for motor 6 on a Hex copter.
Board seems OK as it works in Betaflight.
Similarly, no servo output on this pad when configured for AIRPLANE.

shellixyz, 2018-03-06:
I can reproduce this issue. On my board the S6 pad is not connected directly to any pin of the MCU so it would explain the issue...

digitalentity, 2018-03-06:
@MATEKSYS - hardware issue?

shellixyz, 2018-03-06:
Scratch that it looks like there is a resistor between the MCU pins and the output pads.

shellixyz, 2018-03-06:
S6 is connected to PA8 via a 200ohm resistor. So the output definition in `target.c` looks good.

flyfisher2, 2018-03-06:
As a work around, I have edited target.c and used a customairplane mix to route the rudder output to pad 5 instead of 6..
It works on the bench but is it safe?

shellixyz, 2018-03-06:
You can do that it is safe. The only thing is that with this mod basically you can't use a different PWM frequency for the ESC and servos. Just use standard 50Hz PWM frequency for the ESC and servo and everything will be alright.

flyfisher2, 2018-03-06:
Thanks for that. I assume that this is because it is sharing Timer 3 with M1?
This is my first foray into FW so I am taking my time with it. 
Hopefully by the time I am ready to fly, this problem will have been fixed.
Thanks again.

shellixyz, 2018-03-06:
Yes it is because timer 3 is shared

flyfisher2, 2018-03-07:
Just noticed that there is a pad 8 on the board.
So, would using the normal target.c (M2 enabled) but using a custom mix to route the elevator to pad 8 be a better work around.

shellixyz, 2018-03-07:
Yes it uses a different timer from the others pins so you wouldn't have the sync ESC/servos PWM limitation.

digitalentity, 2018-03-07:
Interesting findings. @shellixyz any clue why this output doesn't work then?

shellixyz, 2018-03-07:
@digitalentity No idea yet unfortunately. Could it be something special about F7 MCUs ?

digitalentity, 2018-03-07:
It's definitely related to timer driver implementation on F7. Might take a while to figure out.

MATEKSYS, 2018-03-07:
Tested 1.9.0_RC1 & final release just now, no servo signal on S6
Tested 1.8.1 built on Dec.24. It works

[inav_1.8.1_MATEKF722_1224.zip](https://github.com/iNavFlight/inav/files/1787690/inav_1.8.1_MATEKF722_1224.zip)

digitalentity, 2018-03-07:
Found a bug. Good old memory corruption. Not fun to track without SWD. 
@MATEKSYS please consider adding SWD pads on your boards 😉

digitalentity, 2018-03-07:
Fixed by #2891. Closing

PR #2891 (2018-03-07): Fix timer initialization issues with HAL (F7)
Fixes failing PWM timer initialization on F7
Fixes #2872
closed 2018-03-07 by digitalentity