iNavFlight/inav #2874: IST8310 magnetometer

Ralfde, 2018-03-05:
inav 1.9.0 RC2
Omnibus F7V2
The magnetometer IST8310 is not found on the I2C (RX3 and TX3)
GPS is found

Please check also on Omnibus F4V5

IST8310 works fine now on Omnibus F4V3

shellixyz, 2018-03-05:
Do you know if other magnetometers work on the Omnibus F7 I2C ? I checked and it is enabled for the Omnibus F7 target on the external I2C (I2C2). If it is working on the Omnibus F4 it should also work on the Omnibus F7. Are you sure you disabled all functions on UART3 ?

Ralfde, 2018-03-06:
How to do this? With the F4SD it does detect the Magnetometer automatically, when I select the Magnetometer. Do I have to resource the UART3 to none for the F7V2?

shellixyz, 2018-03-06:
I don't think there is something special for the F7V2. You only need to have no function selected for UART3 in the ports tab.

Ralfde, 2018-03-06:
I didn't select anything on the UART port tab for sure.

Ralfde, 2018-03-07:
IST 8310 is found with Betaflight 3.4 test firmware (just show some strange X-values) with Omnibus F7V2, but not with inav. I checked the resource pin configuration: Seem to be correct.
B10: I2C2 SCL
B11: I2C2 SDA
Please check again, where the problem could be. Used the new 1.9.0 release from yesterday.

shellixyz, 2018-03-07:
I can't see any driver for the IST8310 in Betaflight. What is your compass detected as ?

```
# In Betaflight source code master branch from today
% ls -1 drivers/compass                                                                                                                              compass_ak8963.c
compass_ak8963.h
compass_ak8975.c
compass_ak8975.h
compass_fake.c
compass_fake.h
compass.h
compass_hmc5883l.c
compass_hmc5883l.h
compass_qmc5883l.c
compass_qmc5883l.h

% grep -rn 8310 drivers/compass
<nothing>
```

I can myself only check if I see something wrong in the code and not really do any testing because I don't own any IST8310 mag nor an Omnibus F7 FC

Ralfde, 2018-03-07:
Maybe it's not included yet. It's a test code from:
https://github.com/McGiverGim
But he does have a problem with the X-axle values.
Maybe you can exchange code and find out why the IST8310 does not work with F7V2 and maybe you can see why the betaflight code does have wrong X-axle values.
The problem is, that he does have also no hardware:
https://github.com/betaflight/betaflight/issues/5386

shellixyz, 2018-03-07:
Please check if it is working with the latest development code: 
[inav_1.9.1_OMNIBUSF7V2.hex.zip](https://github.com/iNavFlight/inav/files/1790272/inav_1.9.1_OMNIBUSF7V2.hex.zip)

Ralfde, 2018-03-08:
Still no magnetometer found. Connected a HMC5983 and this one is found.
With the Omnibus SD F4V3 board everything works fine and the values look also fine.

Ralfde, 2018-03-08:
We use pull-ups in our circuit, but at betaflight the F7V2 does make 3 I2C errors, before it works correct and with F4SD V3 it have no errors. So this could be a reason. I can test it tomorrow.

shellixyz, 2018-03-08:
You can also try to lower the i2c speed with `set i2c_speed = 100KHZ`

digitalentity, 2018-03-09:
I would be happy to troubleshoot, but I don't have anything with the IST8310 sensor. Other compass sensors (HMC5883 and QMC5883 tested) work perfectly fine on OBF7 V2

Ralfde, 2018-03-09:
Still does not work. When I change from IST8310 to auto detection, then the USB connection is not found any more.

Ralfde, 2018-03-09:
The external MS5611 is also not found on the F7V2.

shellixyz, 2018-03-09:
Did you check the pull-up resistors and the lower I2C clock speed ?

Ralfde, 2018-03-09:
Yes. I tested the lower clock speed, too.
The Pull-ups can not be the reason as Betaflight testsoftware does find the IST8310 after 3 errors with exactly the same hardware.

Ralfde, 2018-03-09:
I really wonder, because the MS5611 on this board is not found, too. Both work on the F4 SD V3 board.

Ralfde, 2018-03-09:
When I tested the HMC5983, it did not work, because the pull ups where missing on this board. When I connected the HMC5983 parallel to the IST8310 the HMC5983 worked, but the IST8310 didn't. So the pull-ups should be okay.

shellixyz, 2018-03-13:
If you connect 2 mags in parallel only one will be detected so since the presence of an HMC5983 is checked first the IST8310 won't be detected.

Ralfde, 2018-03-13:
Sorry, misunderstanding. I tested the IST8310 alone. Just the HMC5983 I could only test together, because my HMC5983 didn't have Pull-ups connected, but the IST8310 does have Pull-ups.
So this was not the problem for the IST8310.
I try to organize samples for you. Will take a while, because I don't have enough right now. But you may tell me where to send it, when I get them.

Ralfde, 2018-03-15:
Could you please check the betaflight code, why they have problems with the X-axle values with IST8310.https://github.com/betaflight/betaflight/issues/5386
or maybe include your calibration code for the IST8310.

qwedsazxc78, 2018-03-20:
hi @Ralfde 
Could you please provide the HW layout for i2c and ist8310? It look like i2c problem in Omnibus F7 but fine in Omnibus F4V3. let us check hw difference.
Also, I can send more sample to you if need. Please send the delivery address to mail alexhsieh@isentek.com.

Ralfde, 2018-03-20:
@shellixyz
Please send him your shipping address for the samples.
Also, I can send more sample to you if need.  @Please send the delivery address to mail alexhsieh@isentek.com.

shellixyz, 2018-03-20:
@Ralfde I sent a sample request

Ralfde, 2018-03-23:
Great. I hope this will make things easier.

Ralfde, 2018-04-06:
Any news in the meantime?

shellixyz, 2018-04-06:
I received the samples but have been a bit busy I will test soon.

digitalentity, 2018-04-23:
@shellixyz any news?

fiam, 2018-04-23:
See #3090

digitalentity, 2018-04-23:
@Ralfde is reporting that the same chip works on Omnibus F4V3 so #3090 is not relevant.

fiam, 2018-04-23:
@digitalentity looks like the chip supports 4 different IC addresses that are configured by pulling down some pins. I suspect that some manufacturers are using different addresses. Are we already checking them all?

digitalentity, 2018-04-23:
@fiam no, we don't check all I2C addresses. That's what I intend to allow in the next PRs

Ralfde, 2018-04-23:
The I2C addressess are not a problem , as Omnibus F4 V3 works.

Ralf

Ralfde, 2018-04-25:
I have another issue with the IST8310, when it is build in flipped CCW 90 degree. If the copter gets some angle, then the compass directions f.e. north and south direction gets wrong. So the calibration seem not work right.
[image]
[image]

If I flip the board to normal direction, then the compass direction is much more stable, when the copter gets an angle.

Could you please check this?

digitalentity, 2018-04-25:
@Ralfde How is the compass chip placed on the board? Is it on top of the board or on the bottom of the board? Can you send a picture?

Ralfde, 2018-04-25:
It's on the top of the board.
[image]

Ralfde, 2018-04-25:
Anyway flying with small angles was possible.

shellixyz, 2018-04-25:
I tested an IST8310 on MATEKF722 and OMNIBUS. Working fine on OMNIBUS but with the MATEKF722 it is auto-detected but values seems wrong (only 0, 1 or -1).

Ralfde, 2018-04-25:
Does it work now also on the Omnibus F7V2 now? Also with larger angles and this build in direction?

shellixyz, 2018-04-26:
It probably doesn't work with any F7 FC. Not sure why yet.

Ralfde, 2018-04-27:
Omnibus F7V2 works with setting I2C address to 0x0C setting and when I delete the other magnetometers and the rangefinder.
Maybe setting the right address is enough. I didn't try.

Ralfde, 2018-04-27:
Still one problem left:
The compass does not hold the direction at higher angles of the copter. I think there maybe something wrong with the calibration routine with the build in direction I have. How could this be checked?

digitalentity, 2018-04-27:
@Ralfde INAV looks for IST8310 at I2C address 0x0C, but there is a PR to change that to 0x0E to match off-the-shelf PX4-compatible units. What address do you prefer for your board?

Ralfde, 2018-04-27:
We would need 0x0C. Could you please check for the 0x0C address first and if it is not there you may change to 0x0E? Or if this is not possible automatically, then make it selectable IST8310 0x0C or IST8310 0x0E.
I found another thing on the Omnibus F7V2. If you choose the MPU6000 and 8kHz PID loop time and 8kHz gyro frequency, then the processor load show >100%, sometimes 1000%. Does it use the 32kHz gyro by mistake or maybe some other bug? With 4kHz it shows <20% processor load the the F4V5 board with MPU6000 and the 8kHz setting is also <50% processor load.

shellixyz, 2018-04-27:
Turns out the IST8310 mag is working fine on the MATEKF722 it was a power issue. IST83xx chips have 2 VCC lines AVCC and IOVCC. Both need to be connected to 3.3V. IST8308 mags are also working fine with the development branch.

irekkinso, 2018-04-27:
I have this mag (ist8310) on this exact FC , wired to 3.3V. and it comes up
as un-available/hardware failure. I've double checked my wiring and set-up
and it's like it's supposed to be. other then the mag the gps unit works
like a charm. 19 or 20 satellites in-doors. in my case I still think the
problem is the I2C address.

Ralfde, 2018-04-27:
Try my changes, then it should find the IST8310.  If you use the other I2C address, then change it in the file.

irekkinso, 2018-04-27:
i'm new to GitHub, at least as far as making changes to anything. I've been
trying to figure out how to do exactly what you suggest. I can get to the
lines in the program and change the address but i'm not sure how you
translate that into something useable. does that make sense. is there a
wiki or GitHub "101" that explains this.

Ralfde, 2018-04-27:
Which target and I2C address do you use?
[inav_1.9.1_OMNIBUSF7V2_20180426.hex.zip](https://github.com/iNavFlight/inav/files/1955768/inav_1.9.1_OMNIBUSF7V2_20180426.hex.zip)
This .hex file is for the  Omnibus F7V2 with IST8310 and address 0x0C

irekkinso, 2018-04-27:
> Sorry, I don't have this equipment. In this case you need to try to change
> it yourself.

l keep working on it. thanks

Ralfde, 2018-07-02:
I updated now to the latest inav 2.0.0 rc02.
Everything works great so far and the copter is flying good.
But the problem, that the compass IST8310 is not exact at higher copter flying angles is still there.
My compass direction is CW 90° flip.
With ardupilot 3.6.0 the compass seem to work without problems, but the flight performance is much better with inav. Could you check again please, if you could improve the compass direction at all possible copter directions? This would be a big improvement. At the moment the copter can not be used for waypoints or so, because of this problem. Just Position hold and coming home with small copter angles does work.
Especially the nose down (flying forward) does make the compass working bad, and this makes the problem for me even more difficult to solve.

digitalentity, 2018-07-02:
@Ralfde please ping me on the email. Chinese guys who manufacture compass chips never got back to me. It would be super if you could send me one of the modules you're testing with. I'm at Switzerland now so I should get the package in a few days.

Ralfde, 2018-07-03:
Dear Konstantin,
to which address should I send it?
Best regards,
Ralf

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iNavFlight/inav/issues/2874#issuecomment-401897888",
"url": "https://github.com/iNavFlight/inav/issues/2874#issuecomment-401897888",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [iNavFlight/inav] IST8310 magnetometer (#2874)",
"sections": [
{
"text": "",
"activityTitle": "**Konstantin Sharlaimov**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@digitalentity",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"iNavFlight/inav\",\n\"issueId\": 2874,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"iNavFlight/inav\",\n\"issueId\": 2874\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/iNavFlight/inav/issues/2874#issuecomment-401897888"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 309378714\n}"
}
],
"themeColor": "26292E"
}
]

digitalentity, 2018-07-03:
> ","avatar_image_url":"

Hi Ralf,
Just sent you a direct email.

Ralfde, 2018-07-03:
Backward tested now with Ardupilot 3.6.0 RC2 again. There the compass works great.
No matter how I turn the copter, the compass direction locks good.

digitalentity, 2018-07-08:
@Ralfde thanks for the hardware. Took me just a few hours to find an issue:

IST8310 is not behaving according to datasheet! It's axis directions are not as documented and in fact are not a right handed coordinate system. Fix pending.

digitalentity, 2018-07-08:
Now IST8310 support is complete and verified. Closing.

closed 2018-07-08 by digitalentity