iNavFlight/inav #2880: Inflight board Adjustments

JonnyFPV, 2018-03-06:
### Board and Version
Omnibus F4 Pro V3
INAV/OMNIBUSF4PRO_LEDSTRIPM5 1.9.0 Mar  5 2018 / 00:22:05 (dc4ef594f)

### Behavior
* _I configured inflight adjustment for the new criuse trottle and board alingnment values. 

if i want to change board alignment for roll it changed values for pitch. And if i want to change pitch it changes roll values in configuration tab.

digitalentity, 2018-03-06:
You probably have yaw board alignment at 90 or 270 deg.

Board alignment values are from the board perspective - if the board is 90deg rotated pitch alignment for aircraft becomes roll alignment for board.

In-flight adjustments (as well as stick commands) are relative to the aircraft for pilot's convenience and firmware takes care of yaw board rotation for you.

shellixyz, 2018-03-07:
I checked again to be sure and it working fine for me. Firmware version 1.9 release and configurator version 1.9.2. The roll adjustment is changing roll angle and pitch adjustment is changing the pitch angle.

digitalentity, 2018-03-07:
I believe this issue is irrelevant. I've also tested and board alignment adjustments work correctly.

closed 2018-03-07 by digitalentity