iNavFlight/inav #2886: Add Spektrum Bidir SRXL

mwsadc, 2018-03-06:
Could Spektrum Bidir SRXL be added as a Serial Receiver Provider, as in Betaflight?

https://github.com/betaflight/betaflight/wiki/Spektrum-SPM4649T-SRXL-Telemetry-setup

fiam, 2018-03-06:
@mwsadc I don't think any of the active contributors have Spektrum gear, so it seems unlikely. Writing code blindly and then asking someone to test it for you is very time consuming (even if you can reuse some code from BF, the RX layer is quite different right now). However, if you can write the code, test it and send a PR it will be gladly accepted.

ninja-zx11, 2018-03-09:
That would be great if someone could please add it.Eagerly waiting for it.Not an expert otherwise i would have tried it at least but absolutely no programming/coding knowledge.I used it on betaflight and it works nicely.Now waiting for it on inav.

closed 2018-05-27 by stale[bot]

mathewpeterson, 2019-08-16:
@SpektrumRC, is this something you would be willing to contribute or sponsor since you added SRXL2 to Betaflight recently?

SpektrumRC, 2019-08-16:
> @SpektrumRC, is this something you would be willing to contribute or sponsor since you added SRXL2 to Betaflight recently?

Sorry not 100% sure what you're asking here? Are you asking if we'd implement Bidi SRXL, implement SRXL2, or providing gear to active developers for testing? 
I have not looked at iNav code much so I'm not sure how much it differs from betaflight now, and how difficult that would make merging the protocol implementations from betaflight to here.

mathewpeterson, 2019-08-16:
@SpektrumRC Yes (inclusive `or`) :)

My ask is to get SRXL (and/or) SRXL2 support into iNav flight firmware.

I only speak on behalf of myself so I have no idea what the active contributors would be willing to do, if anything, if they were provided support from you folks (be it test gear or anything else).

SpektrumRC, 2019-08-16:
@mathewpeterson I (Miguel) have considered taking some time at some point to implement SRXL2 at least since I use iNav in one of my planes. I was going to wait until the SRXL2 implementation that was just merged into betaflight master at least makes it into a release first, then hopefully I can look into how much effort it might take to merge that to iNav. I'm not sure when I'll have a chance to do that though. If one of the active iNav dev's are interested in taking ownership of Spektrum related stuff though, I could possibly look into providing them with some Spektrum gear for testing/verification purposes.

mathewpeterson, 2019-08-16:
Thanks for responding! 

That sounds great to me! Please let me know if there's anything I can help test!

CloudsCapone, 2019-12-01:
I am still not seeing support for SRXL2 with iNav, am I missing it, can it be done in CLI like betaflight?

michalsok, 2020-01-13:
Just adding my vote to see the SRXL added to iNav since telemetry is critical in remote/ semi autonomous flying.
Thank you for considering.

DzikuVx, 2020-01-13:
@SpektrumRC we have more users requesting SRXL and it's really up to you

MiguelFAlvarez, 2020-01-16:
I have added an update to issue #5265 with a link to test builds for SRXL2 support. This does NOT include support for the older Bidi-SRXL.