iNavFlight/inav #2897: Omnibus F4 V2.1: Ibus reciever, GPS red, will not find satellite, no messages

spflyer, 2018-03-09:
I've been trying for days to get the GPS working with my Omnibus F4 V2.1 clone purchased from Banggood. https://www.banggood.com/BF3_1_5-Omnibus-F4-V2-Flight-Controller-STM32-F405-MCU-Integrated-OSD-Built-in-5V-BEC-Current-Meter-p-1143258.html?cur_warehouse=CN  I am using Omnibus F4 Pro as target and the latest version, 1.9 firmware.  Setup has proceeded normally until I get to GPS.  GPS is Beitian B-880, also purchased from Banggood. https://www.banggood.com/UBLOX-NEO-M8N-BN-880-Flight-Control-GPS-Module-Dual-Module-Compass-p-971082.html?cur_warehouse=CN I have my FS IA 6b receiver connected to the board with leads soldered directly to the special pads on the flight controller designated for DSM/IBUS.  They are the pads on the underside of the board next to the SD card holder.  Reciever is designated in Inav as Uart 1 and is working properly.  I have combined the wires from the GPS/compass into one six slot appropriately sized plug and connected it to the plug to the right of the USB port, (SDA,SCL,+5V, GND, TX6,RX6) and the TX/RX leads are inverted per usual.  I've designated UART6 as GPS in INav and set the baud to 115200.  All seems to be wired and configured correctly.  The magnetometer is discovered by Inav and calibrates as it should. Mag is green in pre-arming checks.  I've checked all appropriate boxes to enable the GPS as usual but the GPS fails to acquire any satellites and displays zero messages.  The GPS icon in the header is red and the flight controller emits a series of fast beeps.  I've tried everything to get this to work but have hit a dead end at every turn.  The GPS itself is healthy as I've tried the same GPS from another working model and it has the exact same issue.  Hoping to get this problem resolved.  I'm not a coder but I've built many multirotors.  I should probably mention that this is a hexacopter.  I realize that there's another target that designates M5 as a led strip pad but because I need M5 for one of my motors I can't use that target.  Pretty sure this is firmware issue and I hope we can resolve it.  Many thanks. 

Here is my CLI dump

    

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/OMNIBUSF4PRO 1.9.0 Mar  5 2018 / 00:15:43 (dc4ef594f)

# resources

# mixer
mixer HEX6X

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature GPS
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
[... 501 lines cut ...]
set nav_mc_vel_xy_p = 40
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 0
set tpa_breakpoint = 1500
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

shellixyz, 2018-03-11:
Sounds like your GPS is not wired correctly. Your config looks good. Red GPS icon and fast beeps means the GPS does not communicate with the FC.

spflyer, 2018-03-11:
Well it lights up so it's powered properly and I already tried switching
the TX rx wires and that made no difference. It's wired tx to rx, rx to tx.
I wish it was that simple.  I tried all the usual stuff before I asked for
help.

spflyer, 2018-03-11:
I also tried switching uarts and same issue

spflyer, 2018-03-11:
Compass is working so I haven't crossed any of those wires with the GPS

shellixyz, 2018-03-11:
That's strange. I've got an Omnibus F4 PRO and will soon receive a BN-880 GPS/Mag module. I will see if I can reproduce your issue.

spflyer, 2018-03-11:
Thank you so much. This is a v2.1. It's a bit different from the pro.  But
I'm cautious to see if you can duplicate the issue.

shellixyz, 2018-03-11:
In the meantime if you have got a USB-Serial adapter you could check if your GPS is working by trying to connect to it with the u-blox center PC software.

spflyer, 2018-03-11:
Yes I do have an adapter and I'll try that.

shellixyz, 2018-03-11:
> This is a v2.1. It's a bit different from the pro

I could be wrong but I think if it has a current sensor then it is really a PRO version. Also your config dump show that you are using the OMNIBUSF4PRO target. Anyway it is the same I've got also from Banggood.

spflyer, 2018-03-11:
Ok, the one thing that kind of sets this fc apart is that it has 3 tiny
pads for DSM/Ibus connection.  That's where I have my receiver wires
connected.  I guess the 3 pin spectrum plug can also be used for that, at
least the 2 pins for signal and ground but the power is 3 volts.  Anyway, I
elected to use the pads and receiver is working fine.

spflyer, 2018-03-12:
GPS fires right up in UCenter.  Shows a mess of satellites and my correct position on the map.

spflyer, 2018-03-13:
OK, I've kinda made some progress with this issue.  It appears to be very similar to issue 2567.  I have discovered that I can indeed get the GPS working and I can get the receiver working but I can't get them both to work at the same time.  Right now I have my receiver soldered to the tiny DSM/IBUS  pads next to the sd holder.  In Inav I have Uart 1 selected as serial receiver and the receiver mode designated as serial and Ibus and it's working fine.  But as soon as I connect and enable the gps on uart 6 nothing works any more and I have to reflash the firmware with a full chip erase to start from scratch.  I've also switched to firmware inav_1.8.0_OMNIBUSF4V3_u3i.hex  as was recommended in the thread for issue 2567.  I've tried connecting the gps to the 4 pin connecter but it's the same issue.  I've gone about as far as my experience can take me.  I'll wait until you get the board and gps and hope you can come up with a solution to this.  Thanks,

shellixyz, 2018-03-13:
Well that's a start. You probably made a mistake in the issue number as #2567 is about OSD layouts.

Even though I ordered my board from the same link you referred to in your first comment it doesn't look like we got the same one. I can't locate any pads near the SD card slot that could be used to connect a receiver. The one you can see on the left of the SD card slot is the RSSI input.

[image]

[image]

> In Inav I have Uart 1 selected as serial receiver and the receiver mode designated as serial and Ibus and it's working fine. But as soon as I connect and enable the gps on uart 6 nothing works any more...

Then there is clearly an electrical conflict between the pads you are using for connecting the receiver and UART6. You need to identify precisely the board you got and use the right target. The Omnibus F4 V3 expects the receiver to be connected to UART6 (so you can't use UART6 for the GPS on that board, only the RX) and the Omnibus F4 PRO expects the receiver to be connected to UART1 because they are equipped respectively with an hardware inverter for compatibility with inverted RX serial signal types like S.Bus. You might be able to identify the board by using a multimeter to trace where the RX pad is connected to. Please also take pictures of your board I might be able to identify it. My guess is that you have got an Omnibus F4 V3 and you need to use the dedicated RX pad for the receiver (UART6). Use UART1 or UART3 for the GPS. The issue is that you will need to disconnect the GPS if you want to flash the board if using UART1 and that you won't be able to use a mag if you use UART3 since it shares the same pins for I2C.

spflyer, 2018-03-13:
This is exactly the board that I have.
https://www.banggood.com/BF3_1_5-Omnibus-F4-V2-Flight-Controller-STM32-F405-MCU-Integrated-OSD-Built-in-5V-BEC-Current-Meter-p-1143258.html?cur_warehouse=CN
If you look at the pictures you'll see 3 pads beside the SD slot.  One is
square and the other two are round.  Above they say gnd, +5v ,dsm/Ibus.
That's where I soldered my receiver wires and it works fine.  It gives
Omnibus F4SD as a target however Inav does not have that firmware
available.

spflyer, 2018-03-13:
The issue that I meant to reference is 2576.

spflyer, 2018-03-13:
I've been trying to use the solution outlined in this video but because I'm using ibus I can't use the inverted uart. http://www.fpvtv.co.uk/fpv-equipment/omnibus-f4-pro/2hH7NxOAKns/omnibus-f4-pro-v21-solving-gps-and-receiver-port-wiring-issues-sbus-ppm-inav/

shellixyz, 2018-03-13:
OMNIBUSF4SD for Betaflight is the same as OMNIBUSF4PRO for iNav. You are overcomplicating things unnecessarily. You don't need to use special pads even with I-BUS you can use the RX input. The inverter is controlled by the FC and iNav knows I-BUS is non-inverted so it will disable the inverter.

spflyer, 2018-03-13:
I've actually tried that.  I have leads soldered to the traditional rx
inputs.  I'm not sure however that I tried it with Omnibusf4pro firmware.
I hope it's as simple as this.  Snow day here in RI USA so I've got plenty
of time to experiment. I'll try this and let you know.

spflyer, 2018-03-13:
OK, I did a clean flash of Omnibusf4pro with nothing connected to the board.  Using the traditional reciever connection port the receiver does not work on any uart, but using the leads soldered to the special pads it works fine on Uart 1. Evidently the inversion is not being done by Inav.  I would suppose that's why the special pads were put there.  I'm pretty much back where I started.  This problem seems unique to the Banggood F4 V2.1 board.  I have a Omnibus F3 set up in the traditional way with gps and compass working fine.

shellixyz, 2018-03-13:
It is a global issue with F4 boards since F4 MCUs don't have integrated inverters on UARTs. You are probably right looking at the target.h file it is very likely the inverter on your board is not controllable since it is an OMNIBUSF4PRO, only some have this functionality. The Omnibus board versions are a mess. Now I don't know where those pads you are using for the RX are connected to exactly and since I don't have the same board I can't check but if you can't use the RX at the same time as the GPS on UART6 they must be linked somehow to UART6.

spflyer, 2018-03-13:
OK, I've may have found a work around to this issue.  I have both the receiver and GPS working together.  The only way the receiver will work is on uart 1 soldered to the tiny pads.  What I did is connect the GPS to Uart 3 and there is no conflict, both are working.  However, since uart 3 is the traditional connection for the mag, I now have no way to connect it.  I'm hoping there is a way to connect the mag to another uart.  the only pins designated SDA/SDL are now being used by the GPS RX/TX.  Can the mag be connected to any other pins?

spflyer, 2018-03-13:
I was hoping there was a way to configure another I2C port.  I've got both GPS and receiver working with the GPS on uart 3 but that took away my compass port.

shellixyz, 2018-03-13:
There is no workaround for that unfortunately. The MCU's hardware I2C port uses these pins you can't do anything about that. It is the way this board has been designed.

spflyer, 2018-03-13:
So I guess this board isn't going to do what I need it to do.
https://www.amazon.com/Huawei-Mate-Factory-Unlocked-5-93/dp/B0791VS3N9/ref=sr_1_3/141-9755824-8292426?ie=UTF8&qid=1520859962&sr=8-3&keywords=huawei+mate+se

spflyer, 2018-03-13:
Sorry about that link.  Not sure how it got in there but I can't edit it

shellixyz, 2018-03-13:
You should be able to edit comment by clicking on the pen in the top right corner of the comment. If you want to use 2 UARTs and also the I2C bus you need to find a way to use UART1 and UART6 but it is possible you won't be able to without removing components from your board. The only way to know is to trace the tracks on your board to know what is connected where through which components.

spflyer, 2018-03-14:
Well,I've already invested too much time in this. I've got other boards capable of doing what I need. I'm disappointed but I'm moving on.

spflyer, 2018-03-15:
OK, I don't give up easily but this F4 board, and other Omnibus clones I've experimented with don't play nice with Ibus. Inav will not uninvert the receiver connection in the traditional connection pads, ppm/sbus. If you connect the receiver to uart 6 you lose the i2c port and are unable to connect a compass.  If you connect the receiver uart 1 and GPS to uart 6 there is a conflict between those two uarts that will not allow both GPS and receiver to work at the same time.  I switched to a Omnibus F3 and it all went together perfectly.  As an aside, if you're building a hex in inav with an Omnibus F3 then you can't use uart 6 for the receiver.  If you do then motors 5 and 6 pwm connections are reassigned and the motors don't show up.  I connected the receiver to uart 1, rx, +5v, gnd, and the GPS to uart 2.  Compass to i2c of course.  Uart 6 must be empty/unassigned for motors 5 and 6 to show up.  Feel free to close this issue unless you have anything to add.  Thanks for your help.

P-I-Engineer, 2018-03-20:
you may have a dead battery in your GPS. My gps is taking forever to connect the first time, like 25 minutes. I measured the voltage on the battery on the module and it self drains in minutes. I have a new battery coming to confirm this.

spflyer, 2018-03-20:
Not a dead battery.  Gps is fine in UCenter and is working fine with an Omnibus F3 board.  Omnibus F4 doesn't play nice with Ibus.

vitec113, 2018-04-03:
@spflyer: i use "Omnibus F4 " and working fine.
I use:
-  ibus connect to  Uart1 (GND, 5V TX1 , RX1 )
- GPS connect to Uart6   (GND, 5V TX6 , RX6 )
-  Compass  connect to i2c  (port share pin Uart3  SCL Tx3/ SDA/RX3  ) 
If you don't connect Compass  from i2c, you can pullup resistors (1K) on I2C bus 
https://www.youtube.com/watch?v=-zaIE-s8aHQ

spflyer, 2018-04-03:
That won't work on the v2.1.  I tried that a million times.

txaukos, 2018-05-03:
I have the Omnibus F4 Pro v2, which I do not know if it's version v2.0 or v2.1. The configuration of Paco is the one I use with INAV with a UBLOX 7N, and it works.
On the other hand, if I use Betaflight with the same wiring, the GPS works but the magnetometer does not recognize it.
Sorry for my bad English.

spflyer, 2018-05-03:
If it's 2.1 it will say so on the underside.  V2 might be different but 2.1 wouldn't work for me.

bansawelite, 2018-05-09:
Totallly agree with @trunglp and @txaukos.
I also have the Omnbus F4 PRO v2.

GPS on Uart 6
Compass SCL/TX3 and SDA/RX3 pins ( Depending on your compass/GPS module if the compass isn't detected automatically on the omnibus pull-up resistors must be added)
Add a 1-10K resistor between VCC and SCL (if compass is not detected and you've enabled the mag) 
Add a 1-10K resistor between VCC and SDA (if compass is not detected and you've enabled the mag)
I use SBUS on UART1 and have not had a problem!

Note:
It may take a couple of attempts to get the settings to stick in inav. You may need to set RX/GPS option in the SETUP screen multiple times to get it to stick.
In addition I had to use OmnibusF4V3 in the firmware target ( OmnibusF4Pro did not work for me maybe a clone issue).

One additional note to users reading this.. In Betaflight, the GPS works on and off temporarily (I believe betaflight firmware has issues with GPS and compass). Inav has worked flawlessly however!
RTH and position hold is a beautiful add on!

closed 2018-06-28 by DzikuVx

r3df1v3, 2018-07-13:
Hi. I can confirm the findings of spflyer, as I had a similar issue with a Omnibus F4 Pro V3  (ledstripM5) and a Beitian BN-220 GPS. Using a PPM receiver connected to the RX pin and the GPS on UART1, the FC would ramdomly disconnect from the GPS module, especially when the Radio TX was being used. The FC would beep rapidly and the configurator showed a red GPS icon and no info in the GPS section.
    I solved it by connecting the GPS on UART3 (I'm not using a mag). That way, everything works  fine.
It seems that the RX pin of UART1, is shared with the PPM pin.
I hope this helps anyone else having this problem.

evildronepilot, 2019-01-31:
same thing..... in my case gps on uart 6 ibus on uart 3 no mag..... lousting gps as soon as I plug in the battery