iNavFlight/inav #2898: EZUHF INAV RSSI

rcabuser, 2018-03-09:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. INAV kit from Banggood https://www.banggood.com/Inav-F3-Deluxe-30_5x30_5mm-Flight-Controller-Integrated-with-M8N-GPS-Compass-Baro-OSD-for-RC-Drone-p-1162872.html?rmmds=myorder&cur_warehouse=CN
_Version of INAV used goes here_1.9

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* When setting up RSSI trough PPM on 4ch. lite EZUHF the RSSI when radio is on shows 100%, when i turn radio off it only drops to around 50%.
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

Nandox7, 2018-03-11:
I suspect you may want to mentioned this here: 
https://www.rcgroups.com/forums/showpost.php?p=36002593&postcount=8650

I also have the same problem with EZIHF RX's.

rcabuser, 2018-03-11:
Is that your post on Rcgroups?

Nandox7, 2018-03-11:
Nop not mine.
But I also use EzUHF and that is annoying.

rcabuser, 2018-03-12:
Agree, very annoying:-(

Nandox7, 2018-03-12:
Ok, looked into the code and the RSSI scaling is a bit messy.

Has I noticed it applies to both PWM and ADC readings but don't see it making much sense to PWM type.
Range is 0 to 255 and that only makes sense when you using ADC, PWM will range from 1000 to 2000.
I'd say that for  PWM reading it would be best to have a min and/or max value setting. MwOSD does that using the calibration step where it record the min and max read value but for iNav we could do it manually using a cli param.

Something like:
rssi_pwm_min
rssi_pwm_max

I'm playing with the code and will test it soon.

shellixyz, 2018-03-12:
Yes the RSSI scaling is also applied for the value coming from RC channels. It makes sense because not all radios send 1000 for 0% signal and 2000 for 100% signal like you are experiencing yourself. The value used from the RC channel is offset it is not the 1000-2000 range that is scaled but 0-1000 which is logical (https://github.com/iNavFlight/inav/blob/master/src/main/rx/rx.c#L454).

What you need is offset the PWM RSSI value sent by your radio by -500 looks like. Or you can change the scaling to a different method like you described with min/max values.

rcabuser, 2018-03-12:
I think this has something to do with failsafe, like it goes back and holds at around 1500 when turning radio off, without going all the way down/up.
When you use the internal module in taranis together with an frsky receiver, you can set up e.g. channel 8 for rssi under "inputs" setting and set the range there but when you use an external module on the radio, the rssi choice is not present under "inputs" :-(. This frustrates the hell out of me. It would have been fantastic to have a possibility to tweak this manually as mentioned above.

Nandox7, 2018-03-12:
Yes that is the way to go.
The current scaling is only useful I'd expect for ADC where the reading may be within a quit small range.

For PWM it should always be within the expected range, if not that code will set the constrain rules, but it misses an offset.

I wonder if it would be easier to just set for both PWM and ADC RSSI a min/max value and have the code do the mapping. This way there is no need for an offset or scaling as it will automatically take 
care of both.

I'll try this later, need to see if I have some RX that used Analog format for the RSSI.

rcabuser, 2018-03-12:
Good work:-) yeah i find the scaling as it is now is not much help in. I'm very excited to hear further on how it goes.

rcabuser, 2018-03-15:
Nandox7, have you been able to do any testing??

Nandox7, 2018-03-16:
Just some basic testing and seems to be working. 
Need to try this outside so that I can get values other than min and max and that will happen only during the weekend.

I'm making a few more changes to the code, but if it works well it will be easy to set, either for RSSI from ADC or PWM.

rcabuser, 2018-03-16:
that's absolutely fantastic, i'm very interested in hearing further:-)

Nandox7, 2018-03-16:
Tried again today and seems ok, at least detects correctly full and no signal, so better than before.
Now I need to attenuate the signal to get middle range values and that is hard!

The damm EzUHF punches throw the wall like a cannon. :)
I don't have attenuator for this frequency so only option is to really do the test outdoors.

rcabuser, 2018-03-16:
Ahh great:-) i really hope it works well with the outdoor test.

rcabuser, 2018-03-19:
Hi! did you manage to do any more testing through the weekend?

Nandox7, 2018-03-20:
Hey not much, snow storm during the weekend and it was too windy to be outside.
I did made some tests indoors but I can't reduce signal strength.

I'm adding now some debug information o that I can to log data and see if I notice small variations of the RSSI that may not translate to a % change.

rcabuser, 2018-03-20:
Ah ok, same crap weather as over here. Please post when you have had the possibility to test this:-)

subiblue2002, 2018-03-23:
can someone help me get my ezuhf 4ch. lite to work with my F4 V2 board?
I cant even get receiver values to register.
thanks.

alfawhaaat, 2018-04-17:
Is there a solution to this yet, I am experiencing the same problem. 100% Rssi -> Turn off EzUHF Tx -> ~56% RSSI...

EnglishTurbines, 2018-04-18:
I just set my EZUHF receiver to use "Link Quality".....not... RSSI.....it works ok....Full signal gives a value around 78.....no signal....zero.....inside iNAV OSD....i never worry about range using EZUHF anyway, not with Dipole on my receiver and SR771 on my TX....:)

alfawhaaat, 2018-04-18:
> I just set my EZUHF receiver to use "Link Quality".....not... RSSI.....

What is the exact difference between those two?
But there must be a possibility to get the exact RSSI values, that you know when something is wrong and you are about to whitness a failsafe...
Something like this:
SET_RSSI_0 = 56
SET_RSSI_1 = 100

JorFru, 2018-04-20:
I take @EnglishTurbines indications, using Link Quality on INAV (PPM ch.8 for me), but for convenience I give a 125% multiplier for taking 100% RSSI on good reception and about 9% RSSI with transmitter powered off.
You can do this with
`set rssi_scale = 125
save`
I'm on 1.9.1

I've seen
`
#define RSSI_SCALE_MIN 1
#define RSSI_SCALE_MAX 255
#define RSSI_SCALE_DEFAULT 100
`
on [here](https://github.com/iNavFlight/inav/blob/dc4ef594fa7c43690301e8f2aa2602c582b79f49/src/main/rx/rx.h)
I'll have a look after.

rcabuser, 2018-04-21:
Good work guys:-) can't wait to try this out.

rcabuser, 2018-04-21:
@StewLG , when getting info on RSSI settings "get rssi" i do not get the 2nd. option you have "RSSI type", do you know why??

StewLG, 2018-04-21:
I have been working on firmware changes that are only present in my version. If you or other users would like to try it, and see if it fixes your problems with RSSI, I'd be glad to make you a binary. 

Just tell me your target (https://github.com/iNavFlight/inav/wiki/Supported-boards) and I will be happy to get some feedback. I offered it here because I think the changes will help those struggling with EZUHF (and probably other RSSI).

rcabuser, 2018-04-21:
Ahh ok:-) i sure would love to try it:-)

StewLG, 2018-04-21:
Again, tell me your target (OMNIBUS, MATEK405, etc) and I'll make a binary available.

rcabuser, 2018-04-21:
Yeah of course, it's omnibus.

rcabuser, 2018-04-22:
Where will the binary be available?

rcabuser, 2018-04-25:
Hi @StewLG, when will the binary be available? I am eager to test it out:-)

exeswiss, 2018-07-04:
Could someone fix the problem? I also have an EzUHF receiver and the same problem.

jaames74, 2018-07-15:
same issue here. Inav2.0RC4, spracingF3, frsky X8R with pwm-to-anagog-converter.
X8R rssi is 0-3.3V tested with multimeter. Shows as 32% to 100% in Inav configurator.

fiam, 2018-07-16:
@exeswiss I think no one from the core team has an EzUHF, so we're waiting for @StewLG 's changes.

jaames74, 2018-07-16:
@fiam this issue is not EzUHF specific. Analog RSSI does not work properly on Inav. See #3607

fiam, 2018-07-19:
Fixed in #3613

commit c34a408f (2018-07-16): Add rssi_offset, apply offset and scale to all protocols

This allows any possible received RSSI to be mapped to [0, 100],
no matter the range and offset of value emitted by the RX.

Also, apply the RSSI scale and offset to all RSSI sources except
via MSP (via MSP we expect the sender to be well behaved and send
only values in [0, RSSI_MAX_VALUE]).

Fixes #2898
Fixes #3497
Fixes #3607
PR #3613 (2018-07-18): Replace rssi_scale and rssi_invert with rssi_min and rssi_max
rssi_min and rssi_max allow users to specify their RSSI range by
simply introducing the smaller and bigger values they see in the
configurator or OSD, similarly to how rxrange works.

Supersedes #3611
Fixes #2898,  #3497 and #3607
closed 2018-07-19 by fiam