iNavFlight/inav #2901: Add Map mode for OSD

Nandox7, 2018-03-11:
One great feature that is currently mission in the OSD and that is very good for long range flying is the map mode.
MwOSD has more than one type, but having in iNav at least the default radar type would already be very good.

FOX1185, 2018-03-14:
Good mornibg, it would also be useful to have an indication in osd angle to home to facilitate navigation both inside and outside the mission

gmacfpv, 2018-03-23:
Will inav introduce radar on the OSD to show overlay of home location relative to aircraft position? Same as vector basically?

fiam, 2018-03-23:
@gmacfpv We're getting there. First, we need support for multiple OSD layouts controllable by switch, which is almost ready to be added.

Can you post an image of what the radar mode in vector looks like?

JonnyFPV, 2018-03-26:
@fiam look at this http://www.eagletreesystems.com/image/cache/data/vector-screen-legend-small-900x900.jpg

I think it work like a point moves around the center dot relative to the start position/home position.

fiam, 2018-03-30:
@JonnyFPV Thanks for the link. The OSDs we support right now don't allow pixel based access, so we're going for a separate screen for the map and radar modes which can be controlled by a switch in the radio.

Eventually we might add an automatic override of the OSD to show map/radar when we detect another aircraft which is likely to crash with you.

JonnyFPV, 2018-03-30:
Sounds nice :)

gmacfpv, 2018-03-30:
Sounds really nice!

fiam, 2018-05-07:
I'm currently implementing map mode for INAV, there's a small preview here: https://www.youtube.com/watch?v=HqDBzzS3MBU

The arrow indicates the position/orientation of the aircraft and you can also add other indicators on top of it, the map will draw around them changing the scale dynamically as needed. PR is in #3178.

DzikuVx, 2018-06-06:
I think we can close this one

closed 2018-06-06 by DzikuVx

kardon18, 2018-10-09:
Fiam, I like this function.  
But after using it for a while. I have found that the "Direction indicator" and "Scaling indicator" are not absolutely needed.  
And because they are set in fixed positions on the OSD, when the Radar/Map function is selected. They just get in the way of other OSD settings that we what it place in those locations.

Can there be an option in the Radar/Map settings. To remove them if desired? Or be able to at least move them to another location on the OSD?

teckel12, 2018-10-09:
@kardon18 I would suggest creating multiple OSD layouts and controlling them via a switch.  You then can enable the radar layout with more limited information or your standard OSD layout without radar but with additional information.  Multiple OSD layouts are really powerful.  I have them wired to an unused knob on my transmitter.

gmacfpv, 2018-10-11:
I'm with @kardon18 on this. No need for direction and scale. It takes up crucial spots on the OSD where most people put things like voltage rssi. Would be great to use the map/radar feature alongside these metrics without having to move them to an unusual place for them.

teckel12, 2018-10-12:
@gmacfpv I agree as well. Distance is better than scale and direction is a duplication as well. No reason for these duplicate readings to be part of radar.