iNavFlight/inav #2904: Feature Request: AVCS Flight Mode

janbolt, 2018-03-11:
Suggestion for a new flight mode AVCS aka Attitude Lock,
This mode tries to lock the current attitude if there is no stick input.
It is great for aerobatic planes and supported by many stabilization systems.

DzikuVx, 2018-03-11:
And how is that different than Acro mode?

janbolt, 2018-03-11:
Acro mode controls the rate of rotation, AVCS controls the position.
So lets say the plane is rotated by a gust around the longitudinal axis,
acro mode will stop the motion, whereas AVCS brings back the original position.

This is from the manual of the HobbyEagle A3 Super 3, a popular stabilizer featuring AVCS:

Normal mode
The Normal mode (also referred to as Rate mode) is the most basic function of the gyro. It works based on the rotation rate control  of  each  axis  of the  plane. When  operating  in  this mode,  the  gyro  will  only correct currently  occurring rotational movements,  a  momentary  reaction  will  be  applied  to  the  servos  when  the  plane  rotating  on  corresponding  axis,  after rotation the servos will move back to their neutral position as soon as the plane standing still immediately. Normal mode can 
be used  with  nearly  any size and  type  of  airplanes.  It  can effectively improve the stability andprecision of the plane and reduce the stall point specially.

ATTI-LOCK MODE
The Atti-Lock mode is also referred to as the 3D mode or AVCS mode. Different from normal mode, the gyro will perform a permanent correction for rotational movements on each axis constantly . Moving the sticks will make the plane rotating at a certain speed on the corresponding axis, as long as the sticks are released the plane will stop and lock its current position immediately. This operation mode is well suited for practicing basic 3D maneuvers such as hovering o r knife edge, so we also call it 3D flight mode. Since it can help you to lock the attitude of the plane, it’ s also helpful for landing. It  is worth noting 
that  when the gyro is operated in this mode, the servos will keep driving to one side if a specific control input is given, and the servos will not center even if the sticks are released, this is normal.

shellixyz, 2018-03-11:
So if I understand it right it is basically a mix between angle and acro. Acro style control (angular velocity control) but using the angle controller for stability.

janbolt, 2018-03-11:
you can think of attitude lock = angle lock.
The current attitude is locked if you let go the sticks.

OlivierC-FR, 2018-03-13:
I'm not so sure I get the difference with the actual acro mode to be honest.

digitalentity, 2018-03-13:
@janbolt what you are describing is the behavior of properly tuned ACRO mode

digitalentity, 2018-03-13:
The way I see it, ACRO mode definition and implementation in stabilizers is different from what we are used to in INAV. AFAIK, most stabilizers don't have an I-term component and thus aren't able to compensate for prolonged disturbance. So they invented AVCS as a kludge to deal with this issue.

My fellow pilot uses INAV on his 3D plane with great success and without needing anything expect ACRO mode.

janbolt, 2018-03-17:
Generally there are 2 different gyro modes, rate and heading hold.
A (not scientific) explanation can be found here [https://www.youtube.com/watch?v=_xM-SeHxFPI](url)
INav Acro Mode uses rate, AVCS heading hold.

digitalentity, 2018-03-17:
I understand the difference in *logic*, but I can't understand why we need this AVCS mode? Properly tuned ACRO will counteract disturbance just like AVCS mode would.

janbolt, 2018-03-18:
No, Acro Mode controls rate. But that does not mean the rate is zero if you do not touch the sticks.
Every disturbance makes a movement, Acro Mode will stop this movement but that takes a little time.
During this time the plane has moved and will not return to the attitude before the disturbance.
With AVCS it will.

DzikuVx, 2018-03-31:
I second @digitalentity . Properly tuned PID works like this. After all Iterm does what it does

closed 2018-03-31 by DzikuVx