iNavFlight/inav #2905: imu_dcm_*** parameters getting corrupted on MatekF722 board

wilco1967, 2018-03-11:
### Board and Version
# INAV/MATEKF722 1.9.0 Mar  5 2018 / 01:40:37 (dc4ef594f)

### Behavior
Tried to update firmware to 1.9.0 on a Matek F722 board, that previously was running 1.8,1.
Somehow, it doesn't want to keep the parameters imu_dcm_* 
Straight After a download, the defaults values show OK
# set imu
imu_dcm_kp = 2500
Allowed range: 0 - 65535

imu_dcm_ki = 50
Allowed range: 0 - 65535

imu_dcm_kp_mag = 10000
Allowed range: 0 - 65535

imu_dcm_ki_mag = 0
Allowed range: 0 - 65535

Then I paste the first part of the old 1.8.1 dump, upto, but not including the set imu_dcm_ lines....
Before I type save, they are still the same
However, when I do a save, and then call up the imu_dcm lines, I get the following.
# set imu
imu_dcm_kp = 2108
Allowed range: 0 - 65535

imu_dcm_ki = 16384
Allowed range: 0 - 65535

imu_dcm_kp_mag = 2048
Allowed range: 0 - 65535

imu_dcm_ki_mag = 16384
Allowed range: 0 - 65535

Now, when I try to set them back to the defaults, the first one
# set imu_dcm_kp = 2500
imu_dcm_kp set to 2500
Seems initially to work just fine, but after a save, the value is gone again

However, when attempting the 2nd one, 
set imu_dcm_ki = 50
The CLI freezes, and I cannot do anything anymore.
Only the red led lights on the FC continuously is lid.... the blue one, previously blinking, stops blinking (it either stays off, or continously lit)..

Powering off the FC (pulling the usb cable) and reconnecting seems to wake up the FC, but the configurator refuses to connect (the com port shows up just fine) message
No configuration received within 10 seconds, communication failed

I need to close and re-start the configurator, before it will talk again to the FC

Attached the 1.8.1 dump I tried to copy (upto, but NOT including the IMU_ lines).
And the 1.9.0 dump I ended up with (after pasting the rest,)

I know some parameters have changed in 1.9.0, and I will correct them later, but this doesn't seem correct to start with....

This Matek F722 FC has never flow so far... the plane it is supposed to go into is not finished yet, but I do have a flying wing on an OmnibusF4V3 board (still on 1.8.1) , and a tricopter on an Omnibus F3, on 1.9.1). both flying great.

[iNav 1.9.0 MatekF722 DUMP - 20170311.txt](https://github.com/iNavFlight/inav/files/1800588/iNav.1.9.0.MatekF722.DUMP.-.20170311.txt)
[iNav 1.8.1 MatekF722 DUMP - 20170311.txt](https://github.com/iNavFlight/inav/files/1800589/iNav.1.8.1.MatekF722.DUMP.-.20170311.txt)

digitalentity, 2018-03-13:
Confirmed. Only MatekF722 target seems to be affected

digitalentity, 2018-03-13:
All mixers that use servos are affected. Most likely servo initialization somehow messes up the `imuConfig`

digitalentity, 2018-03-13:
Fix pending merge. Closing

PR #2914 (2018-03-14): Fix PWM output allocation
In the case when number of allocated output exceeds MAX_PWM_OUTPUT_PORTS memory corruption occured.

Fixes #2905
closed 2018-03-13 by digitalentity